Bouni/kicad-jlcpcb-tools #295: Plugin does no longer work on OSX with KiCAD 7.0

Bouni, 2023-03-17:
**Describe the bug**
The plugin does not start at all.

**To Reproduce**
Install the latest version on OSX + KiCAD 7.0

**Expected behavior**
The plugin should start

# KiCad Version

```
7.0
```

Bouni, 2023-03-17:
This was reported to me on other channels and I'll put that here for documentation.

Bouni, 2023-03-17:
It seems that #292 makes the plugin fail on OSX. After the user commented  https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/mainwindow.py#L67 it works again.

badrbouslikhin, 2023-03-18:
> @hansfbaier You reported this in #291, can you please test if changing the line 67 in mainwindow.py to `self.app = wx.App()` works for you as well?

I am running into the same issue.
Changing the line 67 in mainwindow.py to `self.app = wx.App()` doesn't fix the issue. There's no output in the Python console but no JLCPCB Tools window does not appear.
Commenting line 67 fixes the issue though.

hansfbaier, 2023-03-18:
@Bouni Yes, replacing that works for me as well.

octave, 2023-03-19:
I have to comment line 67 of `mainwindow.py` as well to make it work with `Kicad 7.0.1` on `macOS 13.2.1`.

shuki25, 2023-03-20:
> I have to comment line 67 of `mainwindow.py` as well to make it work with `Kicad 7.0.1` on `macOS 13.2.1`.

I confirm that this fix works for me on Kicad 7.0 on MacOS Ventura (13.2.1) on Apple Silicon (M2).

JohnVidler, 2023-03-20:
I also confirm that commenting line 67 on my M1-max Mac running Ventura (12.2.1) with KiCAD 7.0 seems to do the trick.

Bouni, 2023-03-20:
So I fear I have to introduce another set of checks for inconsistency between OSes / wx versions etc. This is such a pain in the rear, you cannot imaging how frustration all this is  😞

Bouni, 2023-03-20:
@hansfbaier Could you please test if this PR breaks the plugin for you?
@ OSx users could somebody of you also verify that this allow the plugin to work on OSx?

hansfbaier, 2023-03-20:
@Bouni Yes, ` self.app = wx.PySimpleApp()` works with KiCAD 6.0 / Linux

closed 2023-03-24 by Bouni