Bouni/kicad-jlcpcb-tools #299: unwanted VScore gerber file

lijon, 2023-03-25:
I have some mounting holes on my PCB (and schematic), footprint MountingHole:MountingHole_3.2mm_M3_Pad.
When generating gerbers from your plugin, it created a VScore gerber file with a circle around these mounting holes. Which layer does it grab this from? There's no VScore layer in KiCad.

# KiCad Version

```
Application: KiCad arm64 on arm64

Version: 7.99.0-577-gc65e33ac13, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.3.1
	FontConfig 2.14.0
	libcurl/7.79.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.45.1

Platform: macOS Monterey Version 12.5.1 (Build 21G83), 64 bit, Little endian, wxMac

Build Info:
	Date: Mar 24 2023 01:20:28
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.3
	Curl: 7.77.0
	ngspice: 38
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:

```

Bouni, 2023-03-27:
Its from the User Comments layer: https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L171

That's not very obvious I must admit ... It was introduced via a PR back in August 2022: https://github.com/Bouni/kicad-jlcpcb-tools/pull/161

@truhlikfredy @lijon 

What do you think would be a better layer for v-scores?

lijon, 2023-03-31:
I have always done V-scores on the edge cut layer with JLCPCB, and it "just works".

Bouni, 2023-04-19:
@lijon Sounds not ideal to me, I would go for a dedicated V-Score layer, not sure how to get this right as I never used v-scores so far ...

Bouni, 2023-04-19:
I was wrong 😅 

https://support.jlcpcb.com/article/68-instructions-for-ordering

>  7. Designs about slots/CUT OUT/ millings/ v-cut
**Please kindly make sure that the v-cut lines, cut outs, millings and slots are in the same layer with the board outline.** If it is not in the same layer with the board outline, it will be missed. So please kindly check it before you place your order.If it is missed due to they are not in the same layer with the board outline, we will not responsible for it.(If the slots are to be plated, it needs to be with the drill holes in the same layer, or it will be missed easily).

polemon, 2024-03-14:
> I have always done V-scores on the edge cut layer with JLCPCB, and it "just works".

How do you differentiate a V-Cut line to regular routing when it's on the same layer?
I also order with JLCPCB but I've never used V-Cut lines, but I want to optimized my panelization, etc.

Bouni, 2024-04-02:
I don't know, I have never used v-scores myself. So I have to rely on the experience of others 🤷🏽

z2amiller, 2026-09-26:
This was fixed in #564.

closed 2026-09-26 by z2amiller