Bouni/kicad-jlcpcb-tools #300: Plugin replaced my LCSC part number!

lijon, 2023-03-31:
I was quite surprised to see (after ordereding the boards) that even though I had specified a LCSC = C441760 field in the schematic, the exported BOM from this plugin replaced that with C193080! Which was a much more expensive cap, with worse tolerance (5% instead of 1%). How can I turn off this behaviour?

# KiCad Version

```
Application: KiCad Schematic Editor arm64 on arm64

Version: 7.99.0-577-gc65e33ac13, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.3.1
	FontConfig 2.14.0
	libcurl/7.79.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.45.1

Platform: macOS Monterey Version 12.5.1 (Build 21G83), 64 bit, Little endian, wxMac

Build Info:
	Date: Mar 24 2023 01:20:28
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.3
	Curl: 7.77.0
	ngspice: 38
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
```

markusdd, 2023-04-16:
That is what you entered in the database. It will tell you in the log that it takes the entry from the database and ignores the schematic. Make sure your entry in the table fits what you have in the schematic and double-check your BOM before ordering.

Bouni, 2023-04-19:
@markusdd is rigth. We decided to use LCSC numbers from the project db over the ones from the schematic because of the missing way to write back into the schematic.

Here's an example:

[image]

You can see the message in the logging window at the bottom.

A way to solve this is to use the remove LCSC number button on the right hand side.
That deletes the number from the database. Close the plugin and re-open the plugin and you'll see that the number from the schematic is copied into the project db.

[image]

Bouni, 2023-04-19:
Maybe we find a better solution for this behavior, suggestions are welcome!

lijon, 2023-04-20:
Sorry for the late reply.
Yeah, I think I probably played around in the UI trying to learn it, and accidentally changed the db for this component without realizing it. The log was flooded with "Part X overridden from database" or similar, one for each part.

Personally I prefer to have a single source for my part numbers, which is the schematic. I guess a better place for a plugin to pick LCSC parts would be in the schematic editor, but I know it doesn't support plugins yet.

zyeborm, 2023-04-28:
Yeah, got bitten by this too. I saw I had copied and pasted a resistor without updating the part number. Updated the part number then went to send it off. Only at the last moment did I see it was still the wrong part.

Why do we want to *ignore* the LCSC part # from the schematic? I'd much much rather take the information from the schematic as the source of truth. As Lijon suggested. Like by all means use the DB if the schematic *isn't* populated, but if the schematic has data in it, that should update the DB not be ignored. 

As far as default behaviours go, ignoring the schematic, or ignoring differences seems like the wrong default.

Perhaps a tick box or button to "load from schematic" and/or a big warning (at the end of the log roll) about there being a difference detected? If you want to live with a difference between SCH and plugin/squelch the warning about it then another checkbox column "Force Plugin Part #" might do the job? To toggle between SCH and DB partnumbers?

Is there an avenue for a bounty to add the SCH writeback or something similar to PCBnew?

It's probably a difference of workflow that exposes this.
At the moment I am using the LCSC website (or https://yaqwsx.github.io/jlcparts/) to search for part#'s and I put them in the schematic by typing for resistors etc or using easyeda2kicad to generate part and footprint already for more complex things, I believe it populates the LCSC part# there too. 

Future features/workflows to solve this.
I think perhaps when eeschema gets the python API or somebody feels really keen having the parts selection in there, and the rotation/position/output in the existing pcbnew would be a really great workflow. For double extra bonus points integrating that easyeda2kicad function would make the whole thing basically one click. I want a MCU (say) hit the "add LCSC part" button search, find and it'll populate the libraries and drop the part into eeschema, with footprints and 3d models attached. Just like EasyEDA, but without all the downsides of easyeda.

A different button could be for just "assign LCSC part" if you want to use the kicad (or your own) part and footprint and just find/attach LCSC numbers.

What would perhaps be nice as a feature is a *global* database with rotations (perhaps saved per part?) and offsets in X/Y. Those aren't going to change per project (I imagine) 

A nice to have might be if the global parts rotation/offset DB could be populated by users and synced, people could then have it pre-populated when they started using it. Of course the download of updated rotations would need to be manually triggered and some sort of gatekeeping done on user modifications making it upstream. 
I'd be down for making a web service to host that side of it if that helps?

(A really nice to have would be JLC populating and maintaining at least the basic parts list, but that's a dream lol)

Bouni, 2023-04-28:
This is a historical thing. I totally agree that the schematic should be the first source of truth.
Maybe we could show some kind of warning in the parts list that indicates that there's a mismatch between schematic and project db 🤔

zyeborm, 2023-04-28:
A warning would be a great start. 
Allowing users to update a part from the schematic would be nice too.

haakonnessjoen, 2023-05-07:
I tried selecting all parts in this plugin. Clicking "Remove LCSC part number", and closed the window. When I clicked on the plugin again, it opened, and said "Part xxxx already in the datbase" for all my components. How do I make it save the database?

Also, in a previous version of the software, I could right click a part, and choose "Paste lcsc number". That is not possible any more. To enter the number, it seems like I have to search it up using the search tool when I already know the lcsc number.

I second what @zyeborm  says. There should be a "Update from schematic" button. If you have updated the part number in the scematic.

BotSpace, 2023-05-17:
+1 to this. I love this plugin but man this was a surprise. I should have double checked the final bom, but I think it is unintuitive for the plugin to overwrite the LCSC number specified in the schematic. I would rather pick every LCSC number myself instead of having the plugin decide. I can see how some users would want to have the plugin try to find LCSC parts for them, but I think that should be an option for them to enable. I think the default should be LCSC number pulled from the schematic if it exists, otherwise no LCSC number in the BOM at all. Will need to replace a few hundred 0603 resistors by hand now :(

haakonnessjoen, 2023-05-18:
@BotSpace I agree with you. I would rather the plugin had an option that the schematic is always the master data holder for LCSC numbers. I just sent an order where all the LEDs was the wrong SMD size, because I had corrected it in the schematics. It's easier to change in the schematics, both for one single part, and for a bunch of parts (using the table editor), so it makes no sense for me to use the plugin for setting LCSC numbers. Again, for some people, and maybe for smaller projects, I would think it is fine to use the plugin, but it would really be much better if it was an optional thing. With the "paste LCSC" submenu that was there before it was manageable. But in the newest version, I am struggling with the plugin, i keep having to delete all LCSC numbers in the plugin, to try to get it to read from my schematic, but it is not always saving the db, it seems, so I have to repeat the process multiple times.

I really hope this could be improved upon, because I really enjoy the simplicity of generating the bom, cpl and gerbers with one button. And to see the LCSC stock of each component.

haakonnessjoen, 2023-05-18:
My proposal for a good user experience, would be that if the LCSC field is not provided from the schematic, you can define it in the plugin, that should probably let the current users that isn't currently bothered by this, happy.

If you however have a LCSC number in your schematic, the plugin database should always use this number by default. If you had another value in the plugin db, or if you try to overwrite it with the plugin, it should just show the plugin's number as red besides the actual number, with a mouse-over tooltip that says "This number is not used, as the schematics number takes precedence, remove the number from the schematic to use the plugin's number".

I think that would solve the problem in both user experiences. As you shouldn't both have numbers in the schematic, and in the plugin. You really should choose one or the other to not make a mess.