Bouni/kicad-jlcpcb-tools #303: Generate output without refill zones

radoslav006, 2023-04-15:
**Describe the bug**
Tool refills out-of-date zones even if one select to not do that.

**To Reproduce**
Steps to reproduce the behavior:
1. Project without refilled zones.
2. Click "Genereate".
3. Tool asks "Zones fills are out-of-date. Refill?".
4. I choose "Continue without refill".
5. After gerbers inspections it turns out that zones are refilled anyway.

**Expected behavior**
Zones shall remain unfilled.

# KiCad Version
```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.1), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 15 2022 13:36:27
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2023-04-19:
That is a bug in KiCAD itself and needs to be fixed on the KiCAD side: https://gitlab.com/kicad/code/kicad/-/issues/13135

radoslav006, 2023-04-19:
I see. I just locally commented out the zone refilling for myself.
Maybe this could be also a solution for your repo; in that way the user has choice to fill the zones or not before launching the plugin. The other option is to detect unfilled zones and provide a custom wx pop-up that will ask user to fill or not. But I am not sure if this could be achieved with KiCad API - btw. is there any sensible API documentation available? I am willing to contribute, but have no knowledge of the KiCad API.

If you ask what a mad man do not refill the zones before production, I can show you what a nice copper balancing can be achieved with this (not connected copper balancing is important from a signal integrity perspective) :)
[image]

In the end, you were aware of the problem, so the issue could be closed then, I suppose.

Bouni, 2023-04-19:
I think we could add another option in the settings that lets the user turn off the auto filling!

radoslav006, 2023-04-19:
Indeed, a good idea.

Bouni, 2023-05-23:
Done!

closed 2023-05-23 by Bouni