Bouni/kicad-jlcpcb-tools #304: JLCPCB part cannot be selected in the plugin although it is available

markusdd, 2023-04-16:
Hi, I just orderered a board and I had to modify my BOM in the order process because I could not enter this part with the help of this (otherwise amazing) plugin: C2937625

I have noticed this with some other parts as well.
The plugins lists all kinds of headers from the manufacturer ZHOURI, but this 1*3 header is not there although it is clearly available and in stock in JLCPCBs order flow. It is not being found no matter what I search for.

I can hardly imagine it is really not there, so what could be the culprit here?

# KiCad Version
```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.1, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 11 2023 03:41:46
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 39
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

markusdd, 2023-04-16:
I should add: yes I have re-downloaded the database again just to make sure.

Bouni, 2023-04-19:
This is most likely due to the fact that JLC added this part after I last time updated the database.
Unfortunately they no longer provide the database for download so I can't update anymore.
But you can enter the part number via the right click menu directly:

[image]

Simply copy the desired number from lets say the JLC parts catalog and paste it via the right-click menu

Csory, 2023-04-25:
@Bouni you mentioned in your reply that "Unfortunately they no longer provide the database for download so I can't update anymore."

Does this mean that the plugin will eventually fade out (as using the active JLC stock database was its core feature)?

Bouni, 2023-04-25:
JLC promised back in december that they will provide API access, which they did. Unfortunately the API at its current state is useless as it features no search and is just a way of fetching all parts paginated in I think 500 items per request.
As my free time is almost zero at the moment I'll give the more time to come up with a proper API.
If that doesn't happen I'll reverse engineer their API which they use for their website and integrate that.

For me the main feature of this plugin is the generation of JLC compatible production files.
In the meantime users can simply search parts in the JLC catalog and copy & paste the numbers into the plugin using the right click menu.

In the future KiCAD will get API support in the schematics (the goal is set for V8 but I don't think that they will manage that) which is the better place to select parts in my opinion.

Maybe I'll split the plugin then into a kind of part selection in the schematic and a production plugin in the pcb editor.

We'll see what the future will bring us 😄

markusdd, 2023-04-25:
Issue understood. Nevertheless I found your search in the plugin super helpful, also displaying the selected part from within KiCAD is great.

Is the '500 page API' accessible publicly?

Maybe I can have a look if it is easy to parse.

Bouni, 2023-04-26:
@markusdd Unfortunately no, they provided me with an API key which I should not distribute. I have no clue how that should work  in the long run but at this point the API is rather useless.
You can check out the API in @yaqwsx jlcparts tool: https://github.com/yaqwsx/jlcparts/blob/master/jlcparts/jlcpcb.py

markusdd, 2023-04-29:
other idea: you already have the database, can't you just update/enrich it with the API access you have? In the sense that you create an update of the database which we then download through the plugin?
This way not every user needs a key.

It would really be a bummer wif this plugin would slowly rot away, the selection of the parts through your search mask is so much better and quicker than JLCPCBs webpage.

yaqwsx, 2023-04-29:
Feel free to use the publicly available database from JLCParts.

markusdd, 2023-04-29:
that one is well known but does not provide the amazing convenience of this plugin integrated into KiCAD.

yaqwsx, 2023-04-29:
The EDIT variant is correct.

markusdd, 2023-04-29:
> The EDIT variant is correct.

That is actually an amazing idea. @Bouni is that feasible?
That would be just plain amazing.

Bouni, 2023-04-29:
That's totally feasible but takes some time which very scares at the moment 😕

markusdd, 2023-04-29:
OK I don't wish to promise anything that I can't deliver but I do in fact know Python pretty well.
So if you need some support I could imagine to contribute.

What I would need to know to have a quick initial look:
Where in the codebase would the hookup be to exchange the hardcoded downloaded DB with what @yaqwsx has?
And @yaqwsx : What, in you opinion, would be the most straightforward way to leverage your DB implementation in this plugin?
Can the download happen from a Python client to store it somewhere locally and then interface with it?

markusdd, 2023-04-29:
I did some brwosing through the code and when I get it right all we would need is replace this:
https://github.com/Bouni/kicad-jlcpcb-tools/blob/592f2ed9f9f120b28994f9f20f32f6b3121cd501/library.py#L350

We basically need a new download function that instead of downloading a ready-made .zip trigger the JLCparts interface.
I woudl even go as far as to say we again write all of this to an equally formated db again, so nothing else has to change.
Part details, datasheets etc all come live from the website as now.

Do you have  some docs about your database format?

yaqwsx, 2023-04-29:
It's an SQLite DB with the following tables: https://github.com/yaqwsx/jlcparts/blob/1a07e1ff42fef2d35419cfb9ba47df090037cc7b/jlcparts/partLib.py#L48-L117. And an example on how to fetch it: https://github.com/yaqwsx/jlcparts/blob/1a07e1ff42fef2d35419cfb9ba47df090037cc7b/.github/workflows/update_components.yaml#L45-L50

markusdd, 2023-04-29:
ok lol so it already is an sqlite db. That is amazing xD

so in the worst case all we must do is transform your format into the format that this plugin expects.

markusdd, 2023-04-30:
@yaqwsx is there a proper pure-python way to re-assemble those multi-part zips? They get created using linux zip command and after an hour of frustrated doc-reading and googling it turns out this tool does not split the filkes but actually inserts offsets etc.

Having to rely on WinRar or 7z as external dependencies to unpack is bad. Why split them in the first place?

How would you suggest to get the db downloaded without engaging in to many external dependencies which might not be cross-platform? Pythons 'zipfile' module from the standard cannot handle these as well.

Also...why is it more tahn 5 Gig deflated. that is so much more than @Bouni s version? Are you story complete pictures or the datasheets in this db?

yaqwsx, 2023-04-30:
I suggest you download it and explore it - it will probably answer most of your questions right away:
- the archive is splitted as it is served from GH pages that have file limit of 100 MB,
- the database is so big as it contains all component metadata from LCSC. Not just part numbers and links to datasheets (like to original JLC table)
- there was already a guy asking me about this, and I suggested to him to just repack the database (e.g., stript LCSC metadata). I am not sure if he pulled it off or not. I suggest looking into other issues or discussions from this project.

markusdd, 2023-04-30:
I already explored it and conversion should be very easy from a data perspective. DB Browser under Windows is great for this.

The issue is really the split though. GNU zip under linux (which is what is used in your script) makes it in a way the files cannot simply be concatenated. And relying on an external dependency like 7z is not good for a KiCad plugin (unless they ship it as part of theyr distro or something.

Do you have a way up your sleeve to unpack those files without installing 7z? (with this it obviously works)

yaqwsx, 2023-04-30:
I don't. But if you plan on stripping the database, I am wondering why you need to do it all in Python, as the stripping will happen in CI?

markusdd, 2023-04-30:
haha ok misunderstanding, that was not originally my plan.
Your plugin also functions locally so my intention would've been to download and do it on the user machine.
But I agree, maybe we can even just build a github pipeline here and fetch the smaller version suitable for the plugin.

markusdd, 2023-05-01:
Ok, so I did something:

```py
#!/bin/env python3

import json
import sqlite3
import zipfile
from pathlib import Path
from zipfile import ZipFile

partsdb = Path("parts.db")

# we want to rebuild a new parts.db, so remove the old one
if partsdb.exists():
    partsdb.unlink()

# connection to the jlcparts sb
conn_jp = sqlite3.connect("cache.sqlite3")

# connection to the plugin db we want to write
conn = sqlite3.connect(partsdb)

# schema creation
conn.execute(
    """
    CREATE TABLE IF NOT EXISTS parts ( 
        'LCSC Part',
        'First Category',
        'Second Category',
        'MFR.Part',
        'Package',
        'Solder Joint',
        'Manufacturer',
        'Library Type',
        'Description',
        'Datasheet',
        'Price',
        'Stock'
    )
    """
)

conn.execute(
    """
    CREATE TABLE IF NOT EXISTS mapping (
        'footprint',
        'value',
        'LCSC'
    )
    """
)

conn.execute(
    """
    CREATE TABLE IF NOT EXISTS meta (
        'filename',
        'size',
        'partcount',
        'date',
        'last_update'
    )
    """
)

conn.execute(
    """
    CREATE TABLE IF NOT EXISTS rotation (
        'regex',
        'correction'
    )
    """
)

# load the tables into memory
res = conn_jp.execute("SELECT * FROM manufacturers")
mans = {i: m for i, m in res.fetchall()}

res = conn_jp.execute("SELECT * FROM categories")
cats = {i: (c, sc) for i, c, sc in res.fetchall()}

res = conn_jp.execute("SELECT * FROM components")
comps = res.fetchall()

conn_jp.close()

# now extract the data from the jlcparts db and fill
# it into the plugin database
rows = []
for c in comps:
    price = json.loads(c[10])
    price_str = ",".join(
        [f"{entry.get('qFrom')}-{entry.get('qTo') if entry.get('qTo') is not None else ''}:{entry.get('price')}" for entry in price])
    row = (f"C{c[0]}",  # LCSC Part
           cats[c[1]][0],  # First Category
           cats[c[1]][1],  # Second Category
           c[2],  # MFR.Part
           c[3],  # Package
           c[4],  # Solder Joint
           mans[c[5]],  # Manufacturer
           "Basic" if c[6] else "Extended",  # Library Type
           c[7],  # Description
           c[8],  # Datasheet
           price_str,  # Price
           c[9],  # Stock
    )
    rows.append(row)

conn.executemany("INSERT INTO parts VALUES(?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?)", rows)
conn.commit()
conn.close()

# compress the result
with ZipFile("parts.db.zip", 'w', zipfile.ZIP_DEFLATED) as zf:
    zf.write(partsdb)
```

This converts the jlcparts db entries into what @Bouni||s plugin expects.
Will try it locally, if it works we can turn this into a PR.

Bouni, 2023-05-01:
@markusdd In my opinion we should go the other way and adjust the plugin that it can use @yaqwsx database without the need to get it into the "right" format!

markusdd, 2023-05-01:
> @markusdd In my opinion we should go the other way and adjust the plugin that it can use @yaqwsx database without the need to get it into the "right" format!

1) now I put in all the work :( ;)
2) in all seriousness: I would advise against that for usability. The database from jlcparts contains a lot of things we do not need and the size difference is massive, also the schema is infavorabel for our purposes because manufacturers and categories are an extra table each and need to be mapped when saerched and accessed.
To download the full jlcparts db you need to download a whopping 8 zip parts, reassemble and then unpack them, which results in a size of over 5.5GB unpacked and a download size of about 400MB.
[image]

If we re-organize the db to your format, the zipped download size shrinks to 180MB, and unzipped size is 1.3GB. So unzipped disk usage is about a factor of 3,just leaving out stuff we really don't need. Also jlcparts database gets split in a way which cannot be reassembled in pure python, and introducing an external dependency like 7z is not nice.
For users with weaker PCs or internet lines this makes a huge difference.

So my proposal would be to build a github pipeline here which runs daily, builts a db compatible with this plugin and offers it for download (just like jlcparts does).

markusdd, 2023-05-01:
Short update: Good news, it works.
I have manually copied over my generated db and I now once again see all parts, including the missing 1*3 pin header which was the reason why I created this issue.

There is one small issue left with the db itself, the filtering for in Stock always returns 0 results with my db.
Seems like a datatype issue or something, although I am inserting the values as integers. Need to investigate.

Bouni, 2023-05-01:
@markusdd You're probably right with you points about the file size ...

Can you somehow share your generated db file so that I can have a look. Maybe I see whats causes your issue with the stock levels

markusdd, 2023-05-01:
I figured that one out already. Stock needs to be stored as string as the in stock check is a string comparison, which is fine, but does not work if stock is in integer.

So that one is fixed in my PR already, database works flawlessly.

I also have the split logic in already and tried reassembling just in pure python, works a treat.

pipeline next. (I have a checklist in the PR)

markusdd, 2023-05-02:
@yaqwsx i need your help here:

https://github.com/markusdd/kicad-jlcpcb-tools/actions/runs/4863968058/jobs/8673211098

I am pretty far along now when it comes to generating the DB. The script works and in this workflow I am downloading your cache, do the transformation, create an artifact etc.
I even now managed, afte activating github pages and setting it to 'github actions' as deployment method to deploy to guthub pages. It also says it successfully fetches the zip-part files which I create and uploads them. But how do I find the URL?
There seems to be a part of this whole Github pages thing I do not understand.

Can you comment on this? I think that is the last piece we are missing. Once that works @Bouni can just take over this PR (it doies not change anything yet for the plugin itself) so that the parts.db zips are generated. Once this works on main in his main repo, I will provide a second PR whcih actually adapts the download function to fetch it directly from the github pages of this project.

@bouni for this to work you need to go to the repo settings and under 'pages' set deployment to github actions, otherwise my new pipeline file wont work.

markusdd, 2023-05-02:
Mmmh...really confusing. @yaqwsx I really need your help here, I do not get what they want. I tried to mimic what you are doing in your repo but I do not seemt o get the concept how stuff is deployed.
https://github.com/markusdd/kicad-jlcpcb-tools/actions/runs/4864875554/jobs/8674601090

markusdd, 2023-05-03:
Ha! It works now, I figured it out.
https://markusdd.github.io/kicad-jlcpcb-tools/parts.db.zip.001

you can now download the zip file from a fixed url (the db has 3 parts 001, 002, 003)

I will finalize the first PR now so that it can be merged, once that pipeline works we can then adapt the download function.

markusdd, 2023-05-06:
Ok @Bouni this issue should be solved with PRs #307 and #310.

Please review, 307 must be merged before 310. For more details see PRs.

I have tested everything with my URL locally and it works well.

Bouni, 2023-05-06:
I'll look into it on Monday, I've no computer where I'm right now

markusdd, 2023-05-06:
> I'll look into it on Monday, I've no computer where I'm right now

no worries, just wanted to let you know. Have fun wherever you are with no PC ;) :D

PR #307 (2023-05-08): add script to convert jlcparts db to plugin style db
Work in progress to resolve #304

Strategy: fetch jlcparts database which is generated daily, convert it to this plugins format to bring down size and enhance usability for the purposes of this plugin.
A github pipeline should then run daily, always serving the latest db, tehn just the download function of this plugin needs adaptation.

- [x] conversion script
  - [x] solve issue that in stock search is somehow not working although numbers are in
  - [x split zip in pure python so that it can be reassembled, Github limits to 100MB file size
 [... cut]
closed 2023-05-08 by Bouni