Bouni/kicad-jlcpcb-tools #308: Not finding part available on both LSCS and JLCPCB

coffeenotfound, 2023-05-02:
**Describe the bug**
The "Assign LSCS number" dialog seems to not be able to find certain parts even though they are available on both LSCS and JLCPCB's assembly service. In this case I have a SX2B12.000F2010F30 crystal but I can't find it. Searching for just "SX2B" in the keyword field returns just one part while JLCPCB and LCSC both have around 17 different variants available

**To Reproduce**
Steps to reproduce the behavior:
1. Design in a SX2B12.000F2010F30
2. Open the "Assign LCSC number" dialog for the part
3. Search for "SX2B12.000F2010F30"

**Expected behavior**
The part being available to choose

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.2, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Apr 15 2023 19:18:27
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON
```

coffeenotfound, 2023-05-02:
Darn it, I somehow missed an existing open issue for this. Closing this as it is already tracked by #307

closed 2023-05-02 by coffeenotfound

markusdd, 2023-05-02:
hi, yes we are on a good track to solve this.

In the meantime, feel free to download this build artifact from my test pipelines for my PR.

If you unpack these .zips you will get an updated parts.db which is very recent from yesterday.

Throw it into your plugin directory for the time being and enjoy the latest parts.

Once we have automated all this you can then once again use the download button from within the plugin.
https://github.com/markusdd/kicad-jlcpcb-tools/suites/12605577466/artifacts/674474304