Bouni/kicad-jlcpcb-tools #316: AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'

basson, 2023-05-25:
Hi

Plugin version latest release. 

Traceback of Python Script Errors
```
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/basson/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/home/basson/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 5, in <module>
    from .mainwindow import JLCPCBTools
  File "/home/basson/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 32, in <module>
    from .library import Library, LibraryState
  File "/home/basson/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/library.py", line 13, in <module>
    import requests
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 95, in <module>
    from urllib3.contrib import pyopenssl
  File "/usr/lib/python3/dist-packages/urllib3/contrib/pyopenssl.py", line 46, in <module>
    import OpenSSL.SSL
  File "/usr/lib/python3/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1553, in <module>
    class X509StoreFlags(object):
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1573, in X509StoreFlags
    CB_ISSUER_CHECK = _lib.X509_V_FLAG_CB_ISSUER_CHECK
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'
```

KiCad version
```
Application: KiCad PCB Editor

Version: 6.0.11-2627ca5db0~126~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.15.0-71-generic x86_64, 64 bit, Little endian, wxGTK, pantheon, x11

Build Info:
	Date: Feb  1 2023 20:48:15
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.87.0
	ngspice: 38
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```
pip3 install -U pyOpenSSL

```
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyOpenSSL in ./.local/lib/python3.8/site-packages (23.1.1)
Requirement already satisfied: cryptography<41,>=38.0.0 in ./.local/lib/python3.8/site-packages (from pyOpenSSL) (40.0.2)
Requirement already satisfied: cffi>=1.12 in /usr/local/lib/python3.8/dist-packages (from cryptography<41,>=38.0.0->pyOpenSSL) (1.15.1)
Requirement already satisfied: pycparser in /usr/local/lib/python3.8/dist-packages (from cffi>=1.12->cryptography<41,>=38.0.0->pyOpenSSL) (2.21)
```
pip3 show urllib3
```
Name: urllib3
Version: 2.0.2
Summary: HTTP library with thread-safe connection pooling, file post, and more.
Home-page: 
Author: 
Author-email: Andrey Petrov <andrey.petrov@shazow.net>
License: 
Location: /home/basson/.local/lib/python3.8/site-packages
Requires: 
Required-by: 
```

Bouni, 2023-06-29:
Oh, I see, Ubuntu still has only 6.0.7 in their repos. That distro is such a pain when it comes to up to date software, unbelievable that is has such a wide adoption rate :roll_eyes:

chmorgan, 2025-03-29:
Closing this one out, v6 is pretty old at this point and bug fixes will have to come in the form of PRs from users of this version.

closed 2025-03-29 by chmorgan