Bouni/kicad-jlcpcb-tools #319: Missing Parts

azeteg, 2023-06-01:
Some missing parts I encountered while using the plugin, all which exist on the JLCPCB parts site:

| LCSC Part | Description     | Category                                          |
|-----------|-----------------|---------------------------------------------------|
| C316054   | 0603ESDA-05N    | TVS/Fuse/Board Level Protection                   |
| C404010   | STM32H750VBT6   | Single Chip Microcomputer/Microcontroller         |
| C701784   | DRV8323SRTAR    | Power Supply Chip                                 |
| C375569   | E6C0606RGBC3UDA | Optocoupler/LED/Digital Tube/Photoelectric Device |

asinghani, 2023-11-28:
I believe I've found a patch for this by adding

```
self.category.SetSelection(wx.NOT_FOUND)
```

after line 111 in `partselector.py` - that seems to force wxwidgets to clear out the Category box.

Bouni, 2023-11-28:
@asinghani Nice find!
Do you want to submit a PR?

z2amiller, 2026-09-26:
This was fixed by [Bouni/kicad-jlcpcb-tools#399](https://github.com/Bouni/kicad-jlcpcb-tools/pull/399) (released in 2024.04.01). On macOS a read-only wx ComboBox always selects its first entry, so the dialog used to open filtered to "ADC/DAC/Data Conversion". [Bouni/kicad-jlcpcb-tools#399](https://github.com/Bouni/kicad-jlcpcb-tools/pull/399) added an "All" entry at the top of the list, which doesn't filter by category, so the dialog now opens on "All" and searches every category. With the wxWidgets 3.2.8 that KiCad 10 bundles, the old list still opens on the first category and the current one opens on "All". Closing, but please reopen if you still see it.

closed 2026-09-26 by z2amiller