Bouni/kicad-jlcpcb-tools #327: Export to schematics does not work with copied components

gollux, 2023-07-16:
I am using this wonderful plugin with KiCad 7.0.6. I noticed that the "Export to schematics" function does not export LCSC numbers for some components. Further investigation uncovered that the problematic components are those which were copied from another schematics. In such cases, KiCad stores component reference (e.g., "U11") for both the original and the current schematics and the plugin probably matches the wrong one.

Bouni, 2023-07-17:
@computergeek1507 Could you take a look at this, please?

computergeek1507, 2023-07-17:
@gollux Can you attach a sample project that has this issue?

gollux, 2023-07-17:
Here it is: https://mj.ucw.cz/x/07a3b7b4a2989b463cb93a2ddab899ed/test-project.tar.gz