Bouni/kicad-jlcpcb-tools #328: Export to schematics does not preserve BOM/POS exclude flags

gollux, 2023-07-16:
I am using this wonderful plugin with KiCad 7.0.6. When I export to schematics, the "Exclude from BOM" and "Exclude from POS" attributes are not exported, so they get overwritten the next time I update the PCB from schematics.

FantomXR, 2023-07-17:
True! I ran into this issue as well!

But the plugin is really awesome :-) A game changer!

Bouni, 2023-07-17:
@computergeek1507 This one is also for you I guess 😅

wohali, 2024-02-14:
Having been bit by this one over and over again, I had a look at what's going on.

The "Exclude from position files" and "Exclude from bill of materials" flags are **footprint properties**. Even if I were go in and plumb the connection between this tool and eeschema, you'd start having DRC errors in eeschema because the footprint no longer matches what is in the library.

If you're willing to accept that limitation -- meaning you will have to ignore that entire class of errors -- I can proceed, but it's not straightforward. Right now, the schematic exporter only adds the LCSC property to symbols in your schematic. It doesn't touch footprints at all. Adding additional properties to the symbol to represent JLCPCB BOM/POS info is going to be problematic, because the tool already has BOM/POS values in the footprint. Keeping those in sync is going to be a fool's errand.