Bouni/kicad-jlcpcb-tools #330: Graphical issue. Can't access all buttons on the right.

FantomXR, 2023-07-17:
See attachement. I can not access all the buttons on the right. I can not scroll down either. I'm on a MacBook Pro 16" running at full resolution. 

I'm on KiCad 7.0.0 and using your latest branch. 

[image]

Bouni, 2023-07-17:
I have to find a way to debug Apple stuff somehow as I have no Mac to test on ....

chmorgan, 2023-07-21:
@Bouni I'm on OSX and wanted to take a look at fixing this. How to get kicad plugin manager to use the local cloned repo for easy dev iteration? How do you quickly iterate during dev?

Bouni, 2023-07-21:
Its actually pretty easy, clone this repo and then: 

```sh
cd ~/Library/Preferences/kicad/scripting/plugins
git clone https://github.com/chmorgan/kicad-jlcpcb-tools.git
```

Make sure you have no version of the plugin installed via PCM.

Ideally you create a feature branch before you start making changes to the code:

```
git switch -c fix-macos-ui
# make your changes
git push -u origin fix-macos-ui
```

When you're done, make PR here on GitHub.

While making changes, the reload plugins function of KiCAD is very handy. I have a custom Keyboard shortcut for it `CTRL+SHIFT+R`. This allows you to change stuff and test it on the fly (the plugin must be closed for reloading but better than restarting KiCad every time)

chmorgan, 2023-07-21:
Ahh perfect, this helps a ton and of course on the feature branch.

chmorgan, 2023-07-21:
Fyi the path ended up being:

/Users/cmorgan/Documents/KiCad/7.0/scripting/plugins

Discovered when trying to figure out why it wasn’t showing up and then clicked on “Reveal plugins folder in finder”.

chmorgan, 2023-07-21:
@Bouni alright so I'm able to edit and reload and looking into the issue. It appears that the toolbar tools added with the 2.0 scale factor (the value of self.scale_factor) are simply too large to fit on the screen. They are quite large for me. Maybe 1.5 is a more natural value. In any case, the toolbar apparently doesn't realize them to fit in the available height, and in fact the wxpython docs don't mention toolbars resizing their elements or scrolling them if there isn't enough room.

We could limit the scaling size to 1.5, or not scale at all, or maybe I'm missing something like adding the toolbar to a scrolling container?

Today is my first day with wxpython/wxwidgets so looking for ideas.

Bouni, 2023-07-21:
I can have a look at it on Monday, don't remember how the scaling works. Just that it was a pain in the rear to get it working 🫤

chmorgan, 2023-07-21:
@Bouni you know, a scale factor of 1 looks perfect here, screenshot attached. Maybe this scaling isn't necessary anymore now that wxwidgets is high dpi aware?

[image]

Bouni, 2023-07-24:
@chmorgan I just checked how we get the `scale_factor`.
Its all in here: https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/helpers.py#L38-L42

Would you mind checking what's the return value of `window.GetDPIScaleFactor()` ?

I guess its 2.0 or something like that on OSX

chmorgan, 2023-07-24:
@Bouni it is 2.0 here.

Looking at how things render it doesn't seem correct though to scale icons by that amount.

chmorgan, 2023-07-24:
[image]

chmorgan, 2023-07-24:
From this temporary code added to mainwindow.py:

```
        self.logbox.write(f'{self.scale_factor}')
        self.logbox.write(f'scale factor {self.window.GetDPIScaleFactor()}')
```

Bouni, 2023-08-08:
@chmorgan @serpent213 Would one of you guys mind testing #354 and see if that fixes the scaling issue?

chmorgan, 2023-08-09:
@Bouni looks good here for me. Agree that it doesn't seem like osx needs the scaling for whatever reason, maybe because wxwidgets handles it somehow.

FantomXR, 2023-08-10:
Let me know as soon as you have a new compiled version. I could give it a try!

Bouni, 2023-08-11:
@FantomXR The latest realease should include the fix, let me know if thats the case

chmorgan, 2024-04-19:
@Bouni would recommend this be closed with the changes to the scaling that I had made.