Bouni/kicad-jlcpcb-tools #335: Generation of BOM and CPL data only for parts that have the LCSC field filled in.

wiserst, 2023-07-19:
I already have the LCSC number filled in for the part before I run the plugin. I need to generate production data (BOM and CPL) only for parts that have the LCSC number filled in.

Would it be possible to add a checkbox for this case?

Bouni, 2023-07-19:
Maybe I could add a option in the settings **only lcsc in BOM**

wiserst, 2023-07-19:
I think that if a specific part (designator) does not have an LCSC number assigned to it, it will not be generated in the BOM and there is no point in generating coordinates for it, because it will not be machine assembled.

I do this manually in the "project.db" database. I delete all records that had a blank lcsc field. Then I run the generation.

Bouni, 2023-07-28:
I work on something but that will take a bit of time ...

wiserst, 2023-07-28:
Thank you for your work!

wiserst, 2023-08-10:
Bouni, thank you so much for your work! It's working exactly as I wanted it to. Good luck for the future!