Bouni/kicad-jlcpcb-tools #338: Would you accept an 'auto search' in the PartSelector dialog?

chmorgan, 2023-07-21:
ie. if any of the drop-down or fields changes (and when the dialog loads) a search is performed? That way the user doesn't have to keep clicking on 'Search'?

Bouni, 2023-07-24:
I'm not sure if that's a good feature. Let say no field is populated and you start typing a value, "10nF" for example, you would trigger 4 queries with a giant result set. For a pulldown it seems a lot better.
Maybe if we redesign the entire search to yield only 25 results or so and kind of paginate the rest.
But my experience with wx tells me that that is a huge pain to get right.

If you want to invest the time, feel free to build a prototype so that we can test it out.
If its better than I expect I might merge it!

chmorgan, 2024-04-18:
Fixed in #348

closed 2024-04-18 by chmorgan