Bouni/kicad-jlcpcb-tools #339: Second board in same directory destroys database of first board

thirtythreeforty, 2023-07-22:
**Describe the bug**
When creating multiple KiCAD projects, or multiple boards, in the same directory, this plugin unconditionally uses the database at `jlcpcb/project.db` for all board files.  If data is already present from a different board, it is immediately destroyed upon opening the plugin.

**To Reproduce**
Steps to reproduce the behavior:
1. Create two .kicad_pro files in the same directory.  (This is a little difficult because KiCAD assumes one directory per project; create them in separate dirs and move them into the same dir.)
2. Add component U1 to both projects.  Give each U1 a different value. Import to the PCB.
3. In the first PCB, assign a JLCPCB part to U1.  Save and close the project.
4. Open the second PCB and open this plugin.  The first project's data is removed.

**Expected behavior**
Ideally I expect the plugin to maintain a separate database for each board file. (Intuitively, I expect the .kicad_sch and .kicad_pcb files to contain _all_ the component data, so the presence of the database is already a little surprising.)  Failing that, if the database is from a different board, I want the plugin to show an error about the situation and refuse to modify the existing database.

# KiCad Version
```
Application: KiCad x86_64 on x86_64

Version: 7.0.6, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.13.1
	HarfBuzz 7.3.0
	FontConfig 2.14.2
	libcurl/8.1.2 OpenSSL/3.1.1 zlib/1.2.13 brotli/1.0.9 zstd/1.5.5 libidn2/2.3.4 libpsl/0.21.2 (+libidn2/2.3.4) libssh2/1.11.0 nghttp2/1.55.1

Platform: EndeavourOS Linux, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: Jul  7 2023 09:07:45
	wxWidgets: 3.2.2 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 8.1.2
	ngspice: 40
	Compiler: GCC 13.1.1 with C++ ABI 1018

Build settings:
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON

```

Bouni, 2023-07-26:
@thirtythreeforty Can you provide a project with multiple boards so we can test possible changes?

thirtythreeforty, 2023-07-29:
[Archive.zip](https://github.com/Bouni/kicad-jlcpcb-tools/files/12207593/Archive.zip)

multiboard-a has parts assigned. Opening jlcpcb-tools in multiboard-b triggers the console message:

```
07-29 11:26:45 com_github_bouni_kicad-jlcpcb-tools.store 240:Part U1 is already in the database but value, footprint, bom or pos values changed in the board file, part will be updated, lcsc overwritten/cleared.
2023.07.29 11:26:45 - DEBUG - update_from_board -  Part U1 is already in the database but value, footprint, bom or pos values changed in the board file, part will be updated, lcsc overwritten/cleared.
```