Bouni/kicad-jlcpcb-tools #341: LCSC attributes not applied to subsheets

thirtythreeforty, 2023-07-23:
**Describe the bug**
With a hierarchical schematic, LCSC attributes are only added to the selected sheet, not sub-sheets.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a sheet with a subsheet. Add a part to this subsheet.
2. Assign a component number in jlcpcb-tools.
3. Export the LCSC part numbers to the main sheet.

**Expected behavior**
I expect all the LCSC part numbers to be applied to the selected .kicad_sch, and all subsheets, recursively, referenced by that sheet.

# KiCad Version
```
Application: KiCad x86_64 on x86_64

Version: 7.0.6, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.13.1
	HarfBuzz 7.3.0
	FontConfig 2.14.2
	libcurl/8.1.2 OpenSSL/3.1.1 zlib/1.2.13 brotli/1.0.9 zstd/1.5.5 libidn2/2.3.4 libpsl/0.21.2 (+libidn2/2.3.4) libssh2/1.11.0 nghttp2/1.55.1

Platform: EndeavourOS Linux, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: Jul  7 2023 09:07:45
	wxWidgets: 3.2.2 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 8.1.2
	ngspice: 40
	Compiler: GCC 13.1.1 with C++ ABI 1018

Build settings:
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON

```

Bouni, 2023-07-24:
@computergeek1507 Please take a look if this is easy to fix ...

Bouni, 2023-07-24:
@thirtythreeforty I hope for real API support for schematics in Kicad V8 (V7.99) soon so this gets a lot better to handle. At the moment its kind of hacky 😅

z2amiller, 2026-09-26:
This should be fixed now. [Bouni/kicad-jlcpcb-tools#506](https://github.com/Bouni/kicad-jlcpcb-tools/pull/506) (released in 2024.09.01) made the schematic export follow sub-sheets on KiCad 8 and later. The KiCad 7 path still wrote only the selected sheet, which is what this report ran into on 7.0.6. [Bouni/kicad-jlcpcb-tools#848 · feat(schematic): add schematic…](https://github.com/Bouni/kicad-jlcpcb-tools/pull/848) replaced both with a single walk through the sheet hierarchy, so on KiCad 7 the export now reaches every sub-sheet too, however deeply nested. That's on main and will ship in the first release after 2026.04.03. Closing, but please reopen if you still see it.

closed 2026-09-26 by z2amiller