Bouni/kicad-jlcpcb-tools #342: After upgrading to 7.0.6 on ubuntu 20.04 and re-installing clicking "Tools->External Plugins->Fabrication Toolkit" just opens a browser window pointing to the path of the output files no dialog JLCPCB tools window

46cv8, 2023-07-23:
**Describe the bug**
Clicking "Tools->External Plugins->Fabrication Toolkit" just opens a browser window pointing to the path of some output files no dialog JLCPCB tools window

**To Reproduce**
1. Installed KiCad 6 from their PPA about 1 year ago and installed your plugin at that time using the PCM
2. Uninstalled everything relating to KiCad 6 yesterday, and added the new KiCad 7 PPA and installed it.
3. Re-installed your plugin using your custom repo and the PCM
4. Attempted to click "Tools->External Plugins->Fabrication Toolkit" and a browser and it showed a progress bar for 0.25 seconds (which I can't read) then opened a browser window with some outputs containing the present timestamp, that are not in a format I'm used to from the prior version. I

**Expected behavior**
I was expecting the JLCPCB Tools dialog to popup letting me customize outputs and parts before I click Generate to create the outputs.

# KiCad Version

```
Application: KiCad x86_64 on x86_64

Version: 7.0.6-7.0.6~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.10.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Ubuntu 20.04.3 LTS, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Jul  7 2023 02:32:25
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_SPICE=ON

```

Bouni, 2023-07-24:
Can you post a screenshot of that window?

46cv8, 2023-07-24:
The progress bar that pops up says "Processing..." and here is a link to the browser tab that it opens up.
https://postimg.cc/PNDXdm10

Bouni, 2023-07-25:
Are you sure that you really use this plugin?

"Tools->External Plugins->Fabrication Toolkit" does seem like something else.

Do you see this icon in the toolbar?

[image]

Bouni, 2023-07-25:
I guess you use this one: https://github.com/bennymeg/JLC-Plugin-for-KiCad

46cv8, 2023-07-25:
*face palm* man your totally right. I didn't even realize there were two plugins. I added your repo then searched for JLCPCB and installed the first plugin. I forgot to select your repo from the dropdown in the PCM before searching. I have your plugin installed again and it's working great. :) Very sorry for wasting your time!

closed 2023-07-25 by 46cv8