Bouni/kicad-jlcpcb-tools #349: Components not JLCPCB rotated

tormyvancool, 2023-07-26:
**Describe the bug**
By importing the production files in JLCPB I expected the components had the correct orientation.
Unfortunately this doesn't occur

**To Reproduce**
Steps to reproduce the behavior:
1. Open the project
2. Click  on the JLCPCB button
3. export
4. Load the Gerber and the  BOM/CPL on JLCPCB
5. See the errors ...

**Expected behavior**
I expect that the CPL file is already corrected based on JLCPCB offsets

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.6, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul  6 2023 04:56:45
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

tormyvancool, 2023-07-26:
> So you have to define a rotation correction for the parts that are not correctly oriented.

When you say "So you have to define a rotation correction for the parts that are not correctly oriented.", how can I do it? because I notice it's a fixed offset and if I can insert it in a way that I have not to manually edit each time the CPL, it will be great

tormyvancool, 2023-07-26:
> Unfortunately there is no magic index where we could lookup the rotation correction ...
> 
> KiCAD has no clear definition where pin 1 should be and to my knowledge JLC define pin 1 depending on how the parts are packaged.
> 
> So you have to define a rotation correction for the parts that are not correctly oriented.

I think I found the way (please consider I installed the plugin today ).  It's required (it's not written into the MD file) just to click on ROTATION and then click on UPDATE ... Thank you for your support

Bouni, 2023-07-26:
Yeah, the documentation could be much better. PRs are highly welcome 😁

tormyvancool, 2023-07-26:
> Yeah, the documentation could be much better. PRs are highly welcome 😁

Yep, I can't edit the read.md file but certainly you can do it rapidly with 2 screenshots :-) I sent the link of this repository to a couple of other ppl which are also interested on this solution.

thisisgregm, 2023-11-06:
I think it depends on the exact package footprint. The "IC" term is very generic. @tormyvancool - if this is still an issue please provide footprint name.
@Bouni - BTW I found some rotation errors myself and I added the corrections to my rotation list in the plugin.
Is there a way to maintain the list here, so other people will benefit from this too. If so how and where shall I report it?

konarik, 2024-04-19:
Thank you for information. Maybe the download button can download also the rotations. It was very unclear that I need to click on UPDATE in rotations. I have looked for it several times to find this thread.