Bouni/kicad-jlcpcb-tools #352: The tool keeps removing lcsc numbers from certain parts in my project

0xFEEDC0DE64, 2023-08-04:
The jlcpcb tool keeps replacing the Cnnnnnnn numbers of capacitors with their symbol names C1, C2, C3, ....

If I update those numbers back to jlc, close kicad, reopen it, its back to C1, C2, C3 again. All other parts work fine.

[image]

Using latest released kicad jlcpcb tools from your repository,

kicad is built from latest git sources on archlinux.

I am expecting that the tool also keeps the Cnnnnnnnnn jlcpart numbers for capacitors too, not only inductors and switches or ICs.

0xFEEDC0DE64, 2023-08-04:
the kicad project I am working at the moment is available on github:

https://github.com/0xFEEDC0DE64/esp-midi-board/tree/main/esp32_midi

Bouni, 2023-08-08:
You're right, I see that too. Maybe the same problem that causes #351

Bouni, 2023-08-08:
@0xFEEDC0DE64 What does the logging window print for lets say capacitor C1?

Bouni, 2023-08-08:
This should be fixed with #356

closed 2023-08-08 by Bouni