Bouni/kicad-jlcpcb-tools #358: Light icons in Dark mode

Bouni, 2023-08-10:
Some OS support dark mode, Windows 11 seems to do not 🫤

So I had to revert #353 

@serpent213 Would you mind submitting a new PR and check if using `.IsUsingDarkBackground()` instead of `.IsDark()` is working for you as well?

For me it works ob Windows 11

 `.IsDark()`  returns `True` and  `.IsUsingDarkBackground()`  returns `False`

[image]

serpent213, 2023-08-10:
> `.IsDark()` returns `True` and `.IsUsingDarkBackground()` returns `False`

Software development is fun, isn't it?  😉

`.IsUsingDarkBackground()` works for me. Opened #359.

Bouni, 2023-08-10:
> Software development is fun, isn't it? 😉

Especially software that involves wx 😅

closed 2023-08-10 by Bouni