Bouni/kicad-jlcpcb-tools #362: Crash on startup for some versions of wx

goosenphil, 2023-08-17:
**Describe the bug**
The `2023.08.02` stable version introduced a bug where it crashes on startup if your wx version does not have certain attributes that are queried.

   
The following error appears on screen after attempting to open:
```
Traceback (most recent call last):

File "/home/phil/.local/share/kicad/6.0/3rdparty/plugins/
com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 28, in Run
  dialog = JLCPCBTools(None)

File "/home/phil/.local/share/kicad /6.0/3rdparty/plugins/
com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 129
__init__
  loadBitmapScaled("fabrication.png’, self.scale_factor),

File "/home/phil/.local/share/kicad/6.0/3rdparty/plugins/
com_github_bouni_kicad-jlcpcb-tools/helpers.py", line 66, in
loadBitmapScaled
  if wx.SystemSettings.GetAppearance().IsUsingDarkBackground(}

AttributeError: type object 'SystemSettings' has no attribute
‘GetAppearance'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install stable version `2023.08.02`  on KiCAD stable 6.0
2. Attempt to open kicad-jlcpcb-tools from the PCB editor and a crash dialog will show up

**Expected behavior**
Program should open normally and ignore dark mode behavior if not present on system.

# KiCad Version

```
Application: KiCad

Version: 6.0.11-2627ca5db0~126~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.14

Platform: Linux 6.2.6-76060206-generic x86_64, 64 bit, Little endian, wxGTK, pop, x11

Build Info:
	Date: Feb  1 2023 20:46:36
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.87.0
	ngspice: 38
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

# Linux version

`cat /etc/lsb-release`

```
DISTRIB_ID=Pop
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Pop!_OS 22.04 LTS"
```

Bouni, 2023-08-17:
Just out of curiosity, why are you still using KiCAD V6?

I think about dropping support for it if not enough users can convince me that there is a valid reason to keep it supported.

goosenphil, 2023-08-18:
> Just out of curiosity, why are you still using KiCAD V6?
> 
> I think about dropping support for it if not enough users can convince me that there is a valid reason to keep it supported.

Hi Bouni, I could definitely run a newer version quite simply by installing it on my machine, was just curious about version support for this project since I didn't see it listed in the documentation.

KiCAD V6 is still widely in use by users of Ubuntu and Debian/Ubuntu-based distros which are quite popular.

[Here](https://packages.ubuntu.com/search?keywords=kicad) you can see the current KiCAD versions that come in the package manager by default. From the [Ubuntu download page](https://ubuntu.com/download/desktop) we can see the currently supported versions of Ubuntu are 22.04 LTS and 23.04.

So from the first page we see these KiCAD versions that ship by default:
```
[jammy (22.04LTS)](https://packages.ubuntu.com/jammy/kicad) (electronics): Electronic schematic and PCB design software [universe]
6.0.2+dfsg-1: amd64 arm64 armhf ppc64el

[lunar (23.04)](https://packages.ubuntu.com/lunar/kicad) (electronics): Electronic schematic and PCB design software [universe]
6.0.11+dfsg-1: amd64 arm64 armhf ppc64el riscv64
```

So I am making an assumption based on it being shipped at that version by default in popular Linux distros, but I could be wrong since the KiCAD website does provide alternative repos the user can add to their package manager to download a newer version.

As for dropping support, perhaps we can add a check which pops up a window indicating that the project is not supported in older versions and encouraging the user to update their KiCAD first? I could submit a pull request if you would want that.

Bouni, 2023-08-21:
@goosenphil You're right about Ubuntu/Debian. They are widely used and still ship KiCAD V6 out of the default repos.
I still consider dropping V6 support as it is a huge burden to keep all Versions running. This means handling a lot of differences all over the code base and so far I have no proper way to test all this.

I'll merge your PR now but I really need to come up with a kind of plan what I'm willing to support as my time is very limited and handling a wide range of OSes and KiCAD versions is not feasible for me 🫤

goosenphil, 2023-08-22:
@Bouni I think it's completely fair to drop support for V6 given that the KiCAD website's installation instructions for Debian/Ubuntu is for V7. Supporting too far back will likely lead to many bug reports which distracts from fixing existing bugs and implementing new features.

When dropping support I would suggest adding to the README.md that V7 is the minimum supported, as well as at a simple version check at the start of the program to inform the user they are attempting to run it on a unsupported version. I can submit a PR for these changes to save you time, please let me know if I should proceed.

Bouni, 2023-08-22:
> as well as at a simple version check at the start of the program to inform the user they are attempting to run it on a unsupported version.

Thats a great idea!

PRs are always welcome, my time is next to 0 at the moment 😓

PR #363 (2023-08-25): Added wx attributes check to prevent startup crash
Fixes #362, where a crash happens on startup for certain versions of wx that do not have `GetAppearance` and `IsUsingDarkBackground` attributes. This issue is only present in stable version  `2023.08.02`.

This fix checks for the existence of the attributes before checking their contents.
closed 2023-08-25 by Bouni