Bouni/kicad-jlcpcb-tools #364: KICAD is down with JLCPCB PlugIn

HaiJin-Yun, 2023-08-28:
I had to give up using the old version of kicad-jlcpcb-tools, which I used to use.
The online link no longer works.

So I had to download the latest version of kicad-jlcpcb-tools from GitHub.

Win 11 + kicad-jlcpcb-tools + Kicad 6.11
Win 11 + kicad-jlcpcb-tools + Kicad 7
Win 10 + kicad-jlcpcb-tools + Kicad 6.11
Win 10 + kicad-jlcpcb-tools + Kicad 7

Under all conditions, KICAD is down.
Is there any way to solve it?

https://github.com/Bouni/kicad-jlcpcb-tools/assets/143373670/6e42d930-5e72-4ebb-8161-9a83009566b0

Bouni, 2023-08-28:
I can reproduce this on my machine (Windows 11 and KiCAD 7) but to be honest I have no idea what happens here.
Does this happen when you use the plugin "normal", I mean not open and close it dozen of times in a row?

HaiJin-Yun, 2023-08-28:
1. Until yesterday, this was not the case when I used the older version of 2022.
2. Search feature suddenly stopped working in 2022 version.
3. You installed the latest version on the same Windows.
4. There is a problem with number 1 on the big PCB that I am working on now.
5. The PCB in the video has this problem in 6 times.
6. My Kicad only has standard installation + teardrops.

Bouni, 2023-08-28:
You mean if using a large pcb you can open/close it just one time to let the crash happen but on smaller pcbs it takes more cycles?

Bouni, 2023-08-28:
I opened an issue with KiCAD in the hope that the developers can help me figuring out what goes wrong. Crashing KiCAD limits my ways to figure this out to almost 0 ☹️

HaiJin-Yun, 2023-08-28:
Yes, that's right.
Big PCB is 1 time.
Small PCB is 6 times.
If you open and close it, KICAD will be terminated.

The old plug in of the 2022 version did well.

HaiJin-Yun, 2023-08-28:
Is the 2022 version of the old plug in not available now?
[mainwindow.zip](https://github.com/Bouni/kicad-jlcpcb-tools/files/12452041/mainwindow.zip)

Bouni, 2023-08-28:
You can find all releases here: https://github.com/Bouni/kicad-jlcpcb-tools/releases

HaiJin-Yun, 2023-08-28:
~ Ver 2023.03 :  DB download not possible, Part search not possible.
 Ver 2023. 05 ~ Ver 2023.08.03  : KICAD is crashing after plugin is opened multiple times (I tried.)

closed 2023-09-11 by Bouni