Bouni/kicad-jlcpcb-tools #367: Save Changes window on PCB Editor when close but no changes are made

ZokiPokiMk, 2023-09-01:
**Describe the bug**
If you open the PCB Editor dont do any change, start the plugin, close the plugin, close the Editor window popups asking to save changes even thou no changes were made.
[image]
Even if you click on Save and kicad saves the file, the file content is not changed.

**To Reproduce**
Steps to reproduce the behavior:
1. Open KiCad
2. Open project
3. Click on PCB Editor
4. Click on JLCPCB Tools image button.
5. Click on  X to close the JLCPCB Tools form.
6. Click on X to close the PCB Editor

**Expected behavior**
No Save Changes? popup should occur when no changes are made.

# KiCad Version
```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.7, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 14 2023 02:42:39
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

Bouni, 2023-09-01:
I can only guess but because we access the kicad_pcb file with the plugin, KiCAD assumes we changed it and asks you if you want to save the changes.

I made a copy of a kicad_pcb file, opened the original file with KiCAD, opened the plugin and saved when it asked me.

If I diff these two files there a no changes at all.

So I guess this is just an inconvenience but nothing to be concerned about.

z2amiller, 2026-09-20:
I, too, have been annoyed by this, and I dug into it. Unfortunately it's not something that the plugin is doing -
KiCad itself dirties the PCB when it launches an action plugin by saving the undo buffers -- this allows any
changes made by the plugin to be ^Z'ed as a group, but just due to Implementation Details, pushing to the
buffer counts as a write and it dirties the state of the board (even if there are no actual changes).

It's probably a relatively small fix to KiCad itself -- but I suspect that the KiCad maintainers are focused on KiCad
11 which is supposed to eliminate this SWIG API entirely.  The IPC versions of plugins don't have this same issue.

closed 2026-09-20 by z2amiller