Bouni/kicad-jlcpcb-tools #368: Plugin no longer uses schematic's LCSC values

bask185, 2023-09-01:
**Describe the bug**
I no longer see all LCSC numbers from the schematic. I give every symbol an extra field which is named JLCPCB Part# and a matching LCSC number. This has always worked but no more.

Now I miss half my LCSC numbers while the symbols still have them

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. go to any symbol in the schematic editor and open the properies box
2. Click on 'add field' and type 'JLCPCB Part#' in the left column and the matching LCSC number in the right col
3. Go to PCB editor and update the board
4. Open the JLCPCB plugin

**Expected behavior**
I should see all LCSC values from the schematic. But I don't.

```
# KiCad Version
Application: KiCad x64 on x64

Version: 7.0.5, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: May 27 2023 02:48:13
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON
```

Bouni, 2023-09-11:
You're right, around 7 month ago we changed the helper function and somehow no longer look for JLC values, just for LCSC.

I'll fix that asap!

closed 2023-09-11 by Bouni

bask185, 2023-09-11:
Bought you a coffee 😉

Bouni, 2023-09-11:
Thank you, highly appreciated 😁