Bouni/kicad-jlcpcb-tools #374: rotation mapping automatically flip the rotation sign if the layer is top or bottom

gionag, 2023-09-20:
If i save a mapping for certain package on the top layer, when i use the same package on the opposite layer, also the sign of the rotation should change.

Should internally adapt the sign if the component is placed on top or bottom

Maybe is a recent change on jlcpcb portal

Bouni, 2023-10-26:
Can you give an example so that I can test it?

dzid26, 2024-01-04:
I see the same problem for my double sided board. 

But see the "sign" is already changed here:
https://github.com/Bouni/kicad-jlcpcb-tools/blob/b764df8f686480bc7b494119e9beed2e891d7c8c/fabrication.py#L94-L112
After removing this condition (i.e. always True), so far I've gotten correct results on both sides.

dzid26, 2024-01-10:
I was able to reproduce it with kicad library footprint.

Example footprint is TO-277A with C895438 - rotation 90deg - top:
[image]
Bottom:
[image]

z2amiller, 2026-09-20:
Fixed by #395

closed 2026-09-20 by z2amiller