Bouni/kicad-jlcpcb-tools #376: alignment issue with imperial drill files, but metric gerbers

Hoernchen, 2023-10-16:
JLC just pointed out that some of my via holes are not centered - I was puzzled, because how would that happen for vias? It's not like I'm manually placing holes in pads when creating vias...

Well, turns out the kicad gerbers are always metric (at least since ~2017) but the jlc pcb tools do not set the metric flag when generating the drill files, so this apparently leads to alignment issues for very small vias and larger boards.

The currently generated mix of metric gerber and imperial drill looks fine in gerbview, but this is something that should be fixed if jlc complains, and the jlc example at https://jlcpcb.com/help/article/362-how-to-generate-gerber-and-drill-files-in-kicad-7 uses metric for the drill, too.

Bouni, 2023-10-26:
Thats strange, I did a lot of PCBs recently and never got complains from JLC. Is it possible that its maybe a problem if the KiCAD settings itself are in inch or something like that?

Bouni, 2023-10-26:
Ok, I looked into this. until now we explicitly set the units of the drill file to imperial:

https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L233C35-L233C35

Which resulted in a .drl file that looked like this:

```
M48
; DRILL file {KiCad 7.99.0-3139-g21adc0aac5} date 2023-10-17T14:06:02+0200
; FORMAT={-:-/ absolute / inch / decimal}
; #@! TF.CreationDate,2023-10-17T14:06:02+02:00
; #@! TF.GenerationSoftware,Kicad,Pcbnew,7.99.0-3139-g21adc0aac5
; #@! TF.FileFunction,Plated,1,2,PTH
FMAT,2
INCH
```

With this Option set to True to get metric units the exact same file will look like this:

```
M48
; DRILL file {KiCad 7.99.0-3139-g21adc0aac5} date 2023-10-26T07:15:05+0200
; FORMAT={-:-/ absolute / metric / decimal}
; #@! TF.CreationDate,2023-10-26T07:15:05+02:00
; #@! TF.GenerationSoftware,Kicad,Pcbnew,7.99.0-3139-g21adc0aac5
; #@! TF.FileFunction,Plated,1,2,PTH
FMAT,2
METRIC
```

So I guess I'll change that to `True` in order to generate metric files.

closed 2023-10-26 by Bouni