Bouni/kicad-jlcpcb-tools #377: Change parts db path

dzid26, 2023-10-19:
**Is your feature request related to a problem? Please describe.**
The database is by default downloaded here
Documents are by default synchronized to the the cloud. It's a Windows thing and it is convenient. Sycnhroinizing plugins and other kicad settings might be convenient too.  Unfortunately, jlcpcb plugin stores 2GB of JLC data in the documents as well. 
`C:\Users\dzidm\OneDrive\Documents\KiCad\7.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\jlcpcb`

**Describe the solution you'd like**
Change default path for storing database.

**Describe alternatives you've considered**
Ability (setting) to move jlc database to some external temp folder. 

**Additional context**
[image]

dzid26, 2023-10-19:
OneDrive also gives the ability to exclude specific uploads based on file extensions. `*.bak` is a no-brainer.   Excluding `.db` is not ideal, but a viable workaround.

Bouni, 2023-10-26:
I agree that we should have a config option for setting the path for the db file. Will look into it soon. About the compression, there is not much I can do and I don't want to put too much effort into a platform specific feature.