Bouni/kicad-jlcpcb-tools #378: No longer see JLCPCB plugin in the PCM

johncohn, 2023-10-23:
**Describe the bug**
I am tryign to reinstall the JLCPCB plugin in the PCM and I no longer see the plugin

**To Reproduce**
* Use 'Manage Repositories' in the PCM to verify  that https://raw.githubusercontent.com/Bouni/bouni-kicad-repository/main/repository.json is in the repository list
* Search for JLCPCB plugin - it is not found

I've been using your wonderful JLCPCB plugin for many months and love it. Tonight I had some problems with JLCPCB part numbers being lost. I went to PCM. and uninstall your JLCPCB plugin, then attempted to reinstall it. I could not find it in the list of available plugins.  I do see the Megadish plugin for JLCPCB.. but I much prefer yours. 

**Expected behavior**
I expected to find the JLCPCB plugin

# KiCad Version

Application: KiCad

Version: (6.0.9-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/8.1.2 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Version 13.6 (Build 22G120), 64 bit, Little endian, wxMac

Build Info:
	Date: Oct 29 2022 19:31:49
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.3
	Curl: 7.64.1
	ngspice: 37
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```
```

mengstr, 2023-10-23:
Today when I installed this plugin for the first time it took me far too long to realise that I have to not only add the repo in the manager , I also have to select the repo before searching for it. (Actually, since there's only one plugin in the repo I didn't have even search site it was the top item in the list  ^__^)
[image]

Bouni, 2023-10-26:
I know that this is a hassle but in order to get the plugin into the official repo, I need JLCPCB to write an email to the KiCAD maintainers that the are OK with that.
I tried this several times an never got the to do so ...

Bouni, 2023-10-26:
@johncohn is this issue solved for you as well?

closed 2023-11-24 by Bouni