Bouni/kicad-jlcpcb-tools #379: Center finding not accurate due to silk screen

dzid26, 2023-10-25:
GetBoundingBox unfortunately takes silk screen into account. 
https://github.com/Bouni/kicad-jlcpcb-tools/blob/eca019d8366539c363e5b429c2e8e5bf4560052a/fabrication.py#L118

As a result for a footprint like this:
[image]
I am getting center a bit too low due to the dot.

dzid26, 2023-10-25:
For Kicad 7.99, this seems to work for some reason without even selecting copper layers:
```
footprint.GetLayerBoundingBox(footprint.GetLayerSet()).GetCenter()
```

Alternatively [combining](https://github.com/yaqwsx/KiKit/blob/d7b219874fa09c4acd06f726e5899b0678de2a13/kikit/common.py#L63C1-L66) boundboxes for pads should work too:

Bouni, 2023-10-26:
@dzid26 are you submitting a PR for this?

Is your approach with combining the bounding boxes of all pads working for V7 and V7.99?

dzid26, 2023-10-26:
I can submit the PR. 

It will work with 7.0. 

Should the same bounding box logic be applied to SMD components though? https://github.com/Bouni/kicad-jlcpcb-tools/blob/eca019d8366539c363e5b429c2e8e5bf4560052a/fabrication.py#L116 It doesn't matter to me, because my crappy footprints (imported from Altium) have this property _unspecified_.  
1.  On one hand, I can imagine people having footprints with SMD property set and the anchor at pin 1 (like mine), so they would benefit from the bounding box calculation.
2. On the other hand, moving anchor relative to footprints can be used to offset component placement in relation to the pads. With the bounding box logic, this would not be possible. However, this would become a moot point if we had a translation correction similar to rotations. 

I tested it with SMD on my old board and it works.

dzid26, 2023-10-26:
In the PR, I prioritized point 1. (I removed ` if get_smd(footprint)`) so that if people have an anchor not in the center, the plugin will help them. 
Point 2. I assume that eventually will be solved with translations - moving anchor on the footprint is a big pain anyway in kicad.

chmorgan, 2025-04-01:
Closing due to inactivity.