Bouni/kicad-jlcpcb-tools #382: Price column is a mess

tormyvancool, 2023-11-03:
---------Add your issue details below----------- -->
When exploring the parts on JLCPCB, the column PRICE is totally a mess.
Please her the screenshot
[image]

What I expect is the real cost of the selected part.
In this moment I have to stop to work on the project and research each part manually as "in the old days" :-/

# KiCad Version

```
Application: KiCad x64 on x64

Version: 7.0.8, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Sep 29 2023 18:44:47
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

Bouni, 2023-11-09:
Thats true!

The problem is, thats how LCSC proviedesprice info via their non official API. The prices are depending on the quantity.
wx is a pain in the rear so I have not yet found a proper way to display the resulting table nicely.
Maybe a tooltip coul be abused ....

Ideas are appreciated

Bouni, 2023-11-24:
I guess I wont fix this as the prices can be seen in the part details window:

[image]

tormyvancool, 2023-11-26:
> I guess I wont fix this as the prices can be seen in the part details window:
> 
> ![grafik](https://private-user-images.githubusercontent.com/948965/285432835-ac1899cc-2e39-4f67-8b6c-0d02875a43fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMTc1NzEsIm5iZiI6MTcwMTAxNzI3MSwicGF0aCI6Ii85NDg5NjUvMjg1NDMyODM1LWFjMTg5OWNjLTJlMzktNGY2Ny04YjZjLTBkMDI4NzVhNDNmYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTEyNlQxNjQ3NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iOWQwNzNjZDRiM2ViOWNkNDE3NWU5NTYzNDU3NmVmNTZlYWQwZWY4M2E0NjY1MGY4ZDA1OTI3ZDEzNzQyNTc2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bL0uZ2wVpHTfMVgOOPx2w2LeiXZoEFbYIvcvvYkXqVY)

Which s way elss immediate than to read it jsu taside. Did you ever used it? because once you use ti, you understand how not practical it is. If I was in ouI could think to work in it and to solve it. Because as you were able to create the new page with the correct price, then you can also show the correct prices into the columns.
If one feature is present: it must work.

tormyvancool, 2023-11-26:
[image]
No I can't support this "I can't" mindset. I will support it when I see that issues are solved.

Bouni, 2023-11-28:
Dear Tormy,

this is a hobby project, I never forced you to give me money, if you decide to do so, great!
If not, no problem.

If I find time to fix an issue I'll do it, but my time is limited and if I fix an issue is up to me.
You need a solution now? Submit Pull request and I merge it if it does not break anything else.

This is how open source work and if you're not ok with it, you might buy an Altium licence and try to convice them to implement features that you want or not happy with. (I don't think that they will act faster or even act at all on the request of a single user).

tormyvancool, 2023-11-28:
> Dear Tormy,
>
> this is a hobby project, I never forced you to give me money, if you decide to do so, great! If not, no problem.
>
> If I find time to fix an issue I'll do it, but my time is limited and [...]

To find a solution to an issue when you have the time to do it it's fine.
To close an issue stating that you **no want** to fix it at all because there is a (cumbersome/nerdy) workaround, ti's something else.
Those are two different approaches very far to each other.

About to submit a pull request:
If I was able to code upon a 3party code, I wasn't even install any 3rd party stuff, but I was making my own package.

AND if aver there was a problem on an already made package, I wasn't even open an issue ticket: just I could Fork it, then Pulling the request. Up there, there is plenty of room to explain to the Master, the reasons of the Pull Request.