Bouni/kicad-jlcpcb-tools #383: Error downloading database

dfear, 2023-11-06:
**Describe the bug**
Cannot download the database.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'JLCPBC Tools'
2. Click on 'Download'
3. See error
Failed to download the JLCPCB database, HTTPSConnectionPool(host='jlc.bouni.de', port=443): Max retries exceeded with url: /parts.zip (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f2fb3841fd0>: Failed to establish a new connection: [Errno -2] Name or service not known'))

**Expected behavior**
The database to be downloaded.

# KiCad Version
```
Application: KiCad x86_64 on x86_64

Version: 7.0.8, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.13.2
	HarfBuzz 8.2.1
	FontConfig 2.14.2
	libcurl/8.4.0 OpenSSL/3.1.3 zlib/1.3 brotli/1.1.0 zstd/1.5.5 libidn2/2.3.4 libpsl/0.21.2 (+libidn2/2.3.4) libssh2/1.11.0 nghttp2/1.56.0

Platform: Manjaro Linux, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: Sep 30 2023 23:32:18
	wxWidgets: 3.2.2 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.83.0
	OCC: 7.6.3
	Curl: 8.3.0
	ngspice: 41
	Compiler: GCC 13.2.1 with C++ ABI 1018

Build settings:
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON

```

thisisgregm, 2023-11-06:
Must be something local at your machine.
I used it a few time yesterday and today on two different machines at two different locations and always downloaded the database with a success.

Bouni, 2023-11-13:
@dfear You most likely have an old version of the plugin installed. The database files comes directly from GitHub nowadays.

Bouni, 2023-11-24:
No feedback, will close this one

closed 2023-11-24 by Bouni