Bouni/kicad-jlcpcb-tools #384: Ability to download and link LCSC datasheet

oliv3r, 2023-11-08:
When having a LCSC part ID supplied to the kicad symbol, the datasheet often is left empty, or an online one with a link that's possibly dead by the time it is needed.

On the LCSC site, we have links to the component datasheet, and while some of them are in chinese-only, it's better then nothing.

I there for request the ability to download and store the datasheet, and link it (relativily) to the project.

E.g. if I have filled in LCSC id C1234, then under kicad project dir subfolder 'docs', the pdf is downloaded, and the field `Datasheet` of the symbol, gets updated with the link `${KICAD_PROJECT_DIR}/docs/<filename>`. This keeps all datasheets close to the project, and available for when they are actually needed.

Bouni, 2023-11-24:
Maybe we should just save the pdf when the button is clicked and open it in the browser locally 🤔

oliv3r, 2023-11-24:
I am not, thank you :)

I use it mostly from the schematics view, which doesn't have this plugin sadly.

My workflow herein, or project organization is, that I like to keep all parts locally and committed to the repo. e.g. docs (or datasheets if you will) contains part datasheets, micro controller manuals etc (yes, this is a lot of duplication on my disk :p) but keeps the repo nicely complete. Downloading datasheets is cumbersome, especially since it is so easy to do.

Having the open button is nice, but ideally, downloading all would be nice :)

btw, I'd even argue, that the functionality should/could be part of https://github.com/uPesy/easyeda2kicad.py and easyeda2kicad should even be integrated in this tool :)

Bouni, 2023-11-24:
As far as i remember @uPesy and I had a conversation about that a while ago, but never atempted to do so 🤷‍♂️

oliv3r, 2023-11-24:
That's a little sad :(

So what I'd really like to see from this tool; as a designer, plugin functionality in eeschm :p so that when sourcing parts during layout, I can just browse the (basic) JLPCB parts/ normal LCSC parts database. easyeda2kicad would then in the background add the right parts, symbol, footprint, 3d part and the pdf. Maybe the pdf functionality is something easyeda2kicad should do (from responsibility point of view).

A nother missing feature in kicad (reported) is that a symbol should/could be quite generic, and then map the pins from the generic symbol, to the lcsc footprint... but I suppose that's a step 2 :)

Bouni, 2023-11-28:
EEschema API is on the horizon but was postponed several times. At the moment its on the V9.0 milestone, so I guess that will take a bit until it actually lands ...

dzid26, 2023-12-10:
> when sourcing parts during layout, I can just browse the (basic) JLPCB parts/ normal LCSC parts database.

@oliv3r you might be interested in my [simple gui](https://github.com/Springer-Electronics/JLC2KiCad_lib/tree/master), short video here: https://github.com/TousstNicolas/JLC2KiCad_lib/issues/58 

Btw, I wonder which is better, **easyeda2kicad** or **JLC2KiCad_lib**.

uPesy, 2023-12-10:
@dzid26 That's really cool to have this kind of GUI to quickly save a part to kicad lib. I'm wondering if it's possible to do it on EEschema window too.  This is something that we had discussed with @Bouni , if it's better to have a complete separeted window GUI from Kicad lunch from a python script.

PR #495 (2024-07-04): Add download datasheet button to partdetails
I've added a "Download datasheet" button to the part details modal:

[image]

This downloads the datasheet to `<project_path>/datahseets`

Fixes #384
closed 2024-07-04 by Bouni