Bouni/kicad-jlcpcb-tools #385: Instructions, Required Fields on Schematic, Example Projects

ForrestErickson, 2023-11-21:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

Your tools look to be very useful and I am pondering what the migration of old project or start of a new project has for schematic field requirements. 
We have existing designs which have fields like this:
[image]

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Can you identify the required fields?  Ie, What are the instructions to set up a new schematic for your tools?
Can you indicate if other fields are allowed or must be eradicated?  
I assume we need a description of the requirements and both simple (trivial) examples (KiCad projects) that meet the requirements and complex examples which have all features.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
None.  

**Additional context**
_Add any other context or screenshots about the feature request here._
For our existing designs what methods of migrating to your tools are required?
How do we reformate our schematic fields with the lease manual effort?

Bouni, 2023-11-24:
You must have a Field that has either LCSC or JLC in its name (case insensitive) which holds a JLC/LCSC number, cor example C2598.

I can see from your Screenshot that Distributor 1 PN is that field, but has the wrong name obviously.

So you can either rename that field or duplicate it and change the name to LCSC PN or JLC PN for example.

ForrestErickson, 2023-11-24:
Thanks for getting back to me and regarding, "So you can either rename that field or duplicate it and change the name to LCSC PN or JLC PN for example."

So I can infer from your answer that the other fields I have do not matter.
Is that correct?

And I can infer that the field order does not matter. 
Is that correct?

Bouni, 2023-11-24:
Exactly!

We only lokk for fields that have either lcsc or jlc in the name, again case insensitive and the first match is used.
If the value does not have a valid LCSC/JLC number C12345 for example we continue and take the next match.

z2amiller, 2026-09-26:
This was more of a request for information - not a feature request.

closed 2026-09-26 by z2amiller

ForrestErickson, 2026-09-26:
> This was more of a request for information - not a feature request.

This is a request for documentation of requirements. 
Show what works and what might not work.

I asked two exact questions:

> So I can infer from your answer that the other fields I have do not matter.
> Is that correct?
> 
> And I can infer that the field order does not matter.
> Is that correct?

Is the answer Yes and Yes?

z2amiller, 2026-09-26:
Yep - other fields and ordering don't matter, we pull from the LCSC fields only and ignore anything else.