Bouni/kicad-jlcpcb-tools #386: How can you search JLCPCB from EESchema?

frohro, 2023-11-24:
I am new to your plugin, and trying to learn how it will integrate with my design style.  I design everything from the schematic and only do the PCB when I like everything in the schematic, so that means I need to search JLCPCB for parts I can use all the time from the schematic.  The web works kind of okay, but as we all know, their online parts search is not as nice as Digikey or Mouser.  I don't seem to be able to use it from EESchema.  If I have footprints selected, I can get them in PCBNew, but I don't have footprints for parts I am not sure about yet.  Without footprints, I can't do anything in PCBNew with the plugin.   What am I missing?

Bouni, 2023-11-24:
Hi,

KiCAD does not have an API for EESchema, just for the PCB editor. They planed to come up with such an API in V7 but postponed it for V8 which I doubt will happen. My guess is that it will land in V9.

The only workaround is to select parts in the PCB editor and use the export to schmeatic function. But be aware that this function is hacky and may not work flawlessly.

frohro, 2023-11-26:
Okay, I get it.  Let's close this until the API is ready for EEschema.
Thanks!

closed 2023-11-26 by frohro