Bouni/kicad-jlcpcb-tools #387: Schematic doesn't get updated

FantomXR, 2023-11-24:
Hey,
again the plugin is great. But its not working properly at my side. I'm working on Mac. 
While the graphical issues have been solved which is great, I now have the problem, that the plugin does not update the schematic.

Here is what I do:
I open up the board (schematic is closed). I open up the plugin. I select a part and toggle BOM. If I open the part inside the board, I can now see that the BOM has been toggled. So far so good. 
Now I do "Export to schematic" to write the new settings to the schematic. 

If I now open the schematic, the new settings have not been overtaken and if I update now the board from the schematic, the new setting gets overwritten by the schematic. 

Same is valid for other things like LSCS part number... I guess this is valid for all settings. 

Is my workflow wrong?

Thanks,
Chris

Bouni, 2023-11-24:
Hi,

the export to schematic function s kind of hacky. KiCAD does not provide a proper API to write to the schematic.
They intended to come up with such an API for V7 but postponed it for V8.
But I doubt that it will make it in V8 as I've not seen even a early alpha of said API.

I cannot inverst time in a feature that is about to be changed anyway soon.

My workflow is to add LCSC numbers manually to the schematic. I totally see that this makes a large part of the plugin kind of useless, but my spare time is so limitid that we will have to live with that until the API lands.

Sorry that I cannot provide a more satisfying solution ...

John-Titor, 2024-04-03:
Thanks for sharing details about the situation here; it makes perfect sense.  Looking at the code, it seems like there are three property names (`LCSC`, `LCSC_PN`, `JLC_PN`?) that are supported, though only `LCSC` is added if they don't exist.
 
Is the correct workaround here to manually add an `LCSC` property to schematic entities with the LCSC part number? That's almost too easy...

chmorgan, 2026-09-06:
PR #793 should resolve the BOM exclusion issue reported here. It updates the schematic's `in_bom` flag during "Export to schematic", so a subsequent "Update PCB from Schematic" should preserve the BOM setting instead of resetting it.

The PR leaves LCSC export behavior unchanged, so the LCSC-number symptom mentioned here may need separate investigation.