Bouni/kicad-jlcpcb-tools #391: Occasional components shift / FP anchor not honored?

thisisgregm, 2023-11-30:
**Describe the bug**
Inconsistent / random components shift, especially non-standard ones like connectors, etc.
Seems like the plugin doesn't honour a footprint anchor (moving an anchor in FP editor, then updating footprints generate the same CPL file while updated footprints should have different components placing than before.

**To Reproduce**
Steps to reproduce the behavior:
1. Download JLC footprint using this tool: https://github.com/TousstNicolas/JLC2KiCad_lib
2. Import to KiCAD library
3. After uploading production files generated using this plugin some components (noticed that for connectors: Micro-USB C2682229 and DC barrel connector C720557) the JLC preview shows the components shifted as shown below (in both examples below J1 is my custom FP, J2 is FP supplied by JLC).
4. I experimented with various FP positions and seems like that doesn't change anything in generated CPL files, so possibly anchor position is not honored by the plugin on certain (unusually shaped?) footprints.

[image]

[image]

**Expected behavior**
A correct components position in CPL file.

# KiCad Version

```
Below version info from my work machine, but the same also happens on 7.0.9 on Linux at home.
The plugin version: 2023.09.01 (latest as of today)

Application: KiCad x64 on x64

Version: 7.0.9, release build

Libraries:
	wxWidgets 3.2.3
	FreeType 2.12.1
	HarfBuzz 8.2.1
	FontConfig 2.14.2
	libcurl/8.4.0-DEV Schannel zlib/1.3

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov  5 2023 19:26:40
	wxWidgets: 3.2.3 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.4.0-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

Bouni, 2023-12-27:
Can you please test again, I just merged PR #381 which might fix this issue.

dzid26, 2024-01-04:
I think #381 is actually a reason of this issue. I was contemplating about this here https://github.com/Bouni/kicad-jlcpcb-tools/issues/379#issuecomment-1780941425.

Bouni, 2024-07-04:
I just tried to reproduce this, my results are not 100% correct but seem to be better than yours:

[image]

The problem is that we don't know for each part what offsets JLC uses, so thats a game of cat and mouse which we will never be able to win.

As this does not mean the board is manufactured wrong, I'll close this as I can not do much about it.

If somebody comes up with a smart idea on how to fix this, re-open or create a new issue or even better a PR!

closed 2024-07-04 by Bouni