Bouni/kicad-jlcpcb-tools #403: plugin crashes on KiCAD 7

SebKuzminsky, 2024-01-18:
I just ran in to the bug from #366.

I'm running KiCad 7 on Debian Bookworm.  I had installed this project using the KiCad PCM, following the instructions in the README. 

When I tried to generate JLCPCB production files I got this error on the console:

```
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Top layer
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Silk top
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Mask top
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Paste top
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Inner layer 1
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Inner layer 2
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Bottom layer
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Silk top
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Mask bottom
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Paste bottom
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted Edges
2024.01.18 16:10:36 - INFO - generate_geber -  Successfully plotted V score cut
2024.01.18 16:10:36 - INFO - generate_excellon -  Finished generating Excellon files
2024.01.18 16:10:36 - INFO - zip_gerber_excellon -  Finished generating ZIP file
Traceback (most recent call last):
  File "/home/seb/.local/share/kicad/7.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 915, in generate_fabrication_data
    self.fabrication.generate_cpl()
  File "/home/seb/.local/share/kicad/7.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 271, in generate_cpl
    for part in self.parent.store.read_pos_parts():
                ^^^^^^^^^^^^^^^^^
AttributeError: 'JLCPCBTools' object has no attribute 'store'. Did you mean: 'Restore'?
```

That used to work for me back on KiCad 6, but fails now that i'm on 7.

I then tried installing the plugin from git (still on KiCad 7), restarted KiCad, and got this different error:

```
2024.01.18 16:26:13 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
2024.01.18 16:26:13 - DEBUG - create_rotation_table -  Create SQLite table for rotations
2024.01.18 16:26:13 - DEBUG - download -  Parts db is split into 3 parts. Proceeding to download...
2024.01.18 16:26:13 - DEBUG - download -  Removing any spurios old zip part files...
2024.01.18 16:26:13 - DEBUG - download -  Download parts db chunk 1 with a size of 76.29MB
2024.01.18 16:26:25 - DEBUG - download -  Download parts db chunk 2 with a size of 76.29MB
2024.01.18 16:26:35 - DEBUG - download -  Download parts db chunk 3 with a size of 59.22MB
2024.01.18 16:26:43 - DEBUG - download -  Combining and extracting zip part files...
Traceback (most recent call last):
  File "/home/seb/.local/share/kicad/7.0/scripting/plugins/kicad-jlcpcb-tools/mainwindow.py", line 556, in populate_footprint_list
    if not self.store:
           ^^^^^^^^^^
AttributeError: 'JLCPCBTools' object has no attribute 'store'. Did you mean: 'Restore'?
```

I'm happy to do any debugging you'd like here.

Bouni, 2024-01-19:
What version of Kicad do you run exactly?
Did you uninstall the old version of the plugin and install it fresh?

heartwerker, 2024-02-16:
Hi there,

same for me with kicad 7 on mac. popup window opens for a few seconds then crashes...

KiCad Version:               7.0.10 
macOS 13.6.1, ARM-64

i attached the report:
[crash-report.txt](https://github.com/Bouni/kicad-jlcpcb-tools/files/14313702/crash-report.txt)

also happy to do further testing in case ...

Bouni, 2024-07-04:
I tried to reproduce on Windows and Linux (Arch) but I can't.
I'll close this, please re-open if this is still a problem.