Bouni/kicad-jlcpcb-tools #409: Filename containing "." is not correctly processed

tjfsteele, 2024-01-25:
**Describe the bug**
If the project filename contains a "." e.g. V1.3 then it gets truncated after the "." so generated files end in V1 instead of V1.3

**To Reproduce**
Steps to reproduce the behavior:
1. Open a project with "." in the filename
2. Open the plugin
3. Generate files
4. See error

**Expected behavior**
Filename should be preserved

# KiCad Version
```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.9, release build

Libraries:
	wxWidgets 3.2.3
	FreeType 2.12.1
	HarfBuzz 8.2.1
	FontConfig 2.14.2
	libcurl/8.4.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov  5 2023 19:26:40
	wxWidgets: 3.2.3 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.4.0-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON
```

tjfsteele, 2024-01-25:
I think this can be fixed by changing fabrication.py in two places as follows

```
        cplname = f"CPL-{self.filename[:-4]}.csv"
        bomname = f"BOM-{self.filename[:-4]}.csv"
```

tjfsteele, 2024-01-25:
Submitted PR https://github.com/Bouni/kicad-jlcpcb-tools/pull/410

Bouni, 2024-01-26:
Hey, thanks for pointing this out!

I've seen your issue and was about to take care of it but realized you already closed the issue.
Please leave it open until the issue is resolved!

tjfsteele, 2024-01-26:
I closed it because I submitted a PR. Happy for you to reopen it if that helps. Thanks for your hard work on this plugin.

Bouni, 2024-01-29:
@tjfsteele Have you seen the review request I made in the PR?

tjfsteele, 2024-01-29:
I can't see any review requests on https://github.com/Bouni/kicad-jlcpcb-tools/pull/410

Bouni, 2024-01-29:
I've replaced it for you. Thanks for your PR anyways 👍🏽

closed 2024-01-29 by Bouni