Bouni/kicad-jlcpcb-tools #417: Assign LCSC number not working

PeeJay, 2024-02-03:
I select a single part and go through the process of assign LCSC number, select the part and click select part, but no LCSC number appears for that part back in the main JLCPCB tools window. If I close it and open the plugin again then the LCSC part number shows up.

If I do the same process with multiple parts selected (like after clicking select alike parts), then no part number is added to any component, even after reopening the plugin.

Plugin is current Git version

# KiCad Version
```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.10, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.2.1
	FontConfig 2.14.2
	libcurl/8.4.0-DEV Schannel zlib/1.3

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 28 2023 21:18:15
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.4.0-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

PeeJay, 2024-02-03:
I just tried the latest release version 2023.09.01 and it all works as expected.

Bouni, 2024-02-05:
I'll check tomorrow

Bouni, 2024-02-23:
Fixed in #423 

Sorry for the delay

closed 2024-02-23 by Bouni