Bouni/kicad-jlcpcb-tools #418: Placement offsets for parts

dhoepfl, 2024-02-05:
**Is your feature request related to a problem? Please describe.**

Some footprints I use result in the parts being placed rotated and with a offset in the JLCPCB placement preview.
For rotation, there is a setting in the plugin but for position offsets, there isn’t (as far as I can tell).

**Describe the solution you'd like**

I’d like to have the option to store a offset that should be used when adding a part to the POS file. Just like the rotation.
Specifying an offset as a relative value (move by dx/dy) would be fine.
Of course there could be other options that simplify the entry:

* Place based on center of footprint (ignoring what layers?)
* Place based on center of pin 1
* Place based on center of all pins
* Place based on top/bottom left/right corner
* …

**Describe alternatives you've considered**

* _Changing the footprints_
  Surely an option for custom footprints but not ideal for standard-parts (e.g. Connector_USB:USB_C_Receptacle_G-Switch_GT-USB-7010ASV vs. C963370)
* _Moving the parts in the placement preview when ordering_
   Works but can easily be forgotten. Especially for parts like the mentioned USB-C connector where the offset is small.

**Additional context**

-/-

Bouni, 2024-02-23:
Thats a known problem for which I've not yet found a good solution. Maybe I'll find one in the near future

BenRoe, 2024-11-14:
Same problem here. A offset would be great.

PR #583 (2025-04-07): Extended rotation correction to also have X/Y offsets.
Implements/Fixes #418

Renamed the rotations.db file to corrections.db to reflect the more general correction approach.
Implemented update logic to convert existing rotation databases to the new correction database (setting the offsets to 0 in this case).
The online corrections file already contains offsets, these are imported.

Some notes about things I am unsure about:

* While implementing this, I noticed that there is a `rotation` table created in the parts.db. This table seems to be never used thus I removed the table creation. (see c [... cut]