Bouni/kicad-jlcpcb-tools #419: Rotation override by component property

lk-davidegironi, 2024-02-12:
Just like using the 'LCSC Part #' property to auto compile the JLC part number, it would be useful if the 'JLCPCB Rotation Offset' were implemented as a Rotation table override for specific board component. Sometime it happens one package has to be rotate in one board and in other not. (I've found this problem with the Potentiometer_Bourns_3296W_Vertical package, as example the C118954 part).

dzid26, 2024-03-07:
> Can you go a bit more into detail why a part has to be rotated in one board but not in the other? That makes no sense to me ....

1. Because there can be multiple footprints with the same name - embedded in PCB file - that have different rotations - this happened to me after importing from Altium.  I didn't bother making 100s of footprints to be all the same orientation. 
2. It is also not guaranteed that JLC parts will be always consistent for a given footprint

Of course for a perfect board / ideal world this shouldn't be an issue.  
In practise, I use https://github.com/bennymeg/JLC-Plugin-for-KiCad for more messy projects which primarily correction per component.

lk-davidegironi, 2024-03-08:
Thank you,

I have tried [JLC-Plugin-for-KiCad](https://github.com/bennymeg/JLC-Plugin-for-KiCad) but I prefer the way this plugin works. To me what can solve this issue is a "JLCPCB Rotation Offset" 'JlcRotOffset'	'JLCRotOffset' field as a parameter, or at least prioritize the "JLCPCB Rotation Offset" parameter if exists. But it has to be implemented as a new features.