Bouni/kicad-jlcpcb-tools #424: KiCAD 8.0 Failed to download JLCPCB DB

Predator2k4, 2024-02-27:
After the installation of KiCAD 8.0 and Installation of your great Plugin, the JLCPCB database download dosn't work.
KiCAD 7.0 is still working.

If you Start the Pluigin in the PCB Editor and press Download, a Popup appears.
Download Error

Failed to download the JLCPCB database, ('Connection aborted.', PremissionError(13,'Premission denied'))

```
Application: KiCad x64 on x64

Version: 8.0.0, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 23 2024 02:24:15
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
```
Thank you

Regards

markusdd, 2024-03-05:
That is extremely odd.

We need more details about your platform and install location etc.

I just tried in my fresh KiCAD 8 install in Windows and it works. I worked with @Bouni to renew the database download etc. a few months back and I cannot reproduce this:
[image]

Bouni, 2024-03-06:
Works for me as well. I guess its a permission problem on @Predator2k4 's machine

Predator2k4, 2024-03-06:
This issue can be closed.
I Tested it again, started KiCad 8 as Admin and the update works fine. 
I checked it with no Admin rights and it wortked now too.
I have nothing changed, so I don't know what was the real issue.

Maybe some communication error with the JLCPCB Server or some hickup on my rig.

I'm sorry for the trouble :(

How I installed KiCad:
I have installed KiCad a drive (M) Not main Partition. The same way KiCad 7.
During the start up of KiCad, it asked, if I want to copy the Library from 7.0, wich i have done.

Predator2k4, 2024-03-06:
No Issue. Issue on user/server side.

markusdd, 2024-03-06:
ok so it's a non-standard install. it might have copied over some wrong ACLs that prevented the writing.