Bouni/kicad-jlcpcb-tools #430: Export to schematic doesn't work on KiCAD v8

whmountains, 2024-03-27:
Export to schematic doesn't work on KiCAD v8.  No error is displayed, but the LCSC field doesn't show up in the schematic.

I notice that the log loads my schematic as if it were kicad v6, not even v7, which is obviously incorrect.  I believe this is caused by the is_nightly check in schematicexport.py.

Thanks for your great work on this plugin!  Other than that one issue, it has basically worked perfectly on v8.

# KiCad Version
```
Application: KiCad Schematic Editor x86_64 on x86_64

Version: 8.0.1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.13.2
	HarfBuzz 8.3.0
	FontConfig 2.15.0
	libcurl/7.79.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.45.1

Platform: macOS Monterey Version 12.5.1 (Build 21G83), 64 bit, Little endian, wxMac

Build Info:
	Date: Mar 14 2024 14:07:38
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.84.0
	OCC: 7.7.2
	Curl: 7.87.0
	ngspice: 42
	Compiler: Clang 14.0.3 with C++ ABI 1002

Build settings:

```

whmountains, 2024-03-27:
I tried replacing `if is_nightly(GetBuildVersion()):` with `if True:` and while that got it to use the KiCad V7 routine, my file was still not updated.  So it looks like a new regex will be needed for kicad v8.

Bouni, 2024-04-02:
@computergeek1507 Can you look into this, please?

computergeek1507, 2024-04-02:
Is it okay if I drop v6 support, what are the versions you would like to support?

bvernoux, 2024-04-02:
It will be amazing to keep v6 if possible (even if it is not supported anymore) as I have lot of old design with KiCad v6 that I do not want to switch to KiCad v7 or v8 as KiCad v6 is rock stable and to avoid some issues ... (I think I'm not the only one)

Bouni, 2024-04-03:
@bvernoux I totally understand and would not delete code that works for v6 if it not stands in the way of any future support, at least for a while.

Bouni, 2024-04-03:
Fixed in #433

Bouni, 2024-04-03:
😬 oh No, I didn't test the function myself before merging.
Hopefully you can fix the issue.

If you want you can email me the file (bouni-kicad-plugin@owee.de) so that I can take a look tomorrow.

computergeek1507, 2024-04-03:
send it to me at scooter_seh(at)yahoo(dot)com

> @Bouni @computergeek1507 thank you so much for your work on this!
> 
> I just tested it, and unfortunately it corrupted my schematic file. To make matters worse I was an idiot and didn't make backups before testing for the first time. I want to help debug this. I'd be happy to privately send the schematic file to either of you to see what went wrong. In the mean time hopefully I will be able to manually edit the file to get back in business.

whmountains, 2024-04-03:
Wow, thank both for replying so quicky!!  I was able to restore the file from an auto-backup.  🎉   I sent the full project file to both of you.  The non-working schematic is suffixed BROKEN in the filename.

Attached below is the diff created when I assign the number from a single part, as well as a screenshot of an error when KiCAD tries to open the resulting schematic.  The only other "change" my difftool can detect is an extra newline at the end of the file.  But I doubt that matters at all.

Staring at that diff, isn't there just a missing parenthesis?

[image]

[image]

computergeek1507, 2024-04-03:
the plugin also makes a copy of the schematic file and adds "_old" before editing it. If it gets corrupts, delete the file and rename the "_old" file to get the original file back

whmountains, 2024-04-05:
Looks like this is fixed in [3959138](https://github.com/Bouni/kicad-jlcpcb-tools/commit/395913897b6ad62124c3da6dc91d95a4d04009e5).  Thanks!