Bouni/kicad-jlcpcb-tools #432: Rotations don't show in the list for the last git clone

gyohng, 2024-04-02:
Hi. Patch to make it work:

```diff
--- a/mainwindow.py
+++ b/mainwindow.py
@@ -610,13 +610,13 @@ class JLCPCBTools(wx.Dialog):
             # First check if the part name mathes
             for regex, correction in corrections:
                 if re.search(regex, str(part[1])):
-                    part[8] = correction
+                    part[8] = str(correction)
                     break
             # If there was no match for the part name, check if the package matches
             if part[8] == "":
                 for regex, correction in corrections:
                     if re.search(regex, str(part[2])):
-                        part[8] = correction
+                        part[8] = str(correction)
                         break
 
             self.footprint_list.AppendItem(part)
```

gyohng, 2024-04-21:
Check this pull request:
https://github.com/Bouni/kicad-jlcpcb-tools/pull/443

chmorgan, 2024-04-25:
@Bouni fix for this is merged into main.

closed 2024-04-25 by gyohng