Bouni/kicad-jlcpcb-tools #438: Part numbers not saved to the schematic or board files?

jmsaltzman, 2024-04-04:
**Is your feature request related to a problem? Please describe.**
I have successfully exported files for PCBA (!!) but I'm not sure how to get the LCSC part numbers in my files independent of the plugin. I see the part numbers in the plugin window, and they exported fine for my JLCPCB order, but I do not see them as properties of components.

**Describe the solution you'd like**
I'd like to know how to get the part numbers as footprint properties (or symbol properties?).

**Describe alternatives you've considered**
I could manually enter them all, and there aren't many, but I imagine there's a way I'm missing.

**Additional context**
I thought "Export to Schematic" would add the part numbers as symbol properties, but when I specified the schematic file, I got this message, and the LCSC's are not in the resulting file:
"2024.04.04 16:47:03 - INFO - _update_schematic -  Added LCSC's to C:\Users\jmsal\Dropbox\KiCad\CH224_breakout_jms_2\CH224_breakout_jms_2.kicad_sch(maybe?)"  <-- "maybe" hmm... :)
It was odd that the save file browser suggested the filename "jms_2.kicad_sch" instead of the longer one; I wonder if my project file's underscores were unexpected.
I had manually added LCSC properties for two footprints in PCBNEW before using your plugin; I wonder if their presence was unexpected and caused an issue. They did appear when I first ran your plugin though-- it was great to see them pipe through.
FWIW I'm running KiCAD 8.0.1 on Windows 10 (latest).

I also want to add: this is AMAZING work and I am so, so impressed! THANK YOU for your thoroughness and generosity!

whmountains, 2024-04-05:
See #430

The issue has been fixed but not yet included in the latest release.

jmsaltzman, 2024-04-06:
Aaaah! Super cool, thank you :)

closed 2024-07-04 by Bouni