Bouni/kicad-jlcpcb-tools #441: Assigns JLCPCB part numbers to handsoldered components

Pancra85, 2024-04-16:
Great plugin!!

My main problem is that it automatically assigns SMD codes to parts that are not SMD.
I am doing a board that has some SMD resistors and caps, as well as THT that I want to solder myself (for debugging,etc)
[image]

Maybe it should assign only on SMD footprints?

Also after I delete them and click on SAVE MAPPINGS, it doesn't get saved when I run the plugin again.

Thanks!

Bouni, 2024-04-17:
Did you assign the LCSC number for R115 in the schematic?
You can use the Remove LCSC number button to delete the assignment:

[image]

Make sure to save the PCB after closing the plugin.

If you don't want a part to be assembled by JLC, use the toggle BOM/POS button. That excludes the part from the BOM as well as the POS file.

[image]

Pancra85, 2024-04-18:
This is what I'm doing right now:
Run the plugin
R115 has a LCSC number assigned on the list, but NOT IN THE SCHEMATIC.
I remove LCSC number and close the plugin
Save PCB
Close PCB, reopen it.
Run plugin and R115 still has a LCSC number assigned

Pancra85, 2024-04-18:
Oh ok, it disappeared now. I guess you need to close and reopen KiCad or the file.
Thanks

closed 2024-04-19 by Bouni