Bouni/kicad-jlcpcb-tools #442: HTTP GET Error

whmountains, 2024-04-19:
I saw the fts5 PR was merged and installed the latest version from the KiCAD PCM to try it out.  But I got this error when opening JLCPCB tools for the first time:

> HTTP GET Error
> Failed to fetch count of database parts, error code 404
> URL was: 'https://bouni.github.io/kicad-jlcpcb-tools/chunk_num_fts5.txt'

The logging area says the following:

> 2024.04.18 19:56:14 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
> 2024.04.18 19:56:14 - DEBUG - create_rotation_table -  Create SQLite table for rotations
> 2024.04.18 19:56:14 - DEBUG - create_rotation_table -  Create SQLite table for rotations

I'm on MacOS 12.5.1, KiCAD 8.0.1, JLCPCB Tools 2024.04.03.

whmountains, 2024-04-19:
It works if edit `library.py` and set `url_stub` back to @chmorgan repo

```python
url_stub = "https://chmorgan.github.io/jlcpcb-db/"
```

Bouni, 2024-04-19:
Oh, that's because the CI job has to run first in order to generate the db files.
It should do so in the next 10 minutes or so anyway so I'll just wait 😅

Bouni, 2024-04-19:
Its 5:00 UTC not 5:00 CEST, so that actually right now. I don't know how long the CI job takes but in a few minutes the files should be ready.

Bouni, 2024-04-19:
Nope, CI job failed. I try to fix it asap.

Bouni, 2024-04-19:
I think this is harder than I thought, I think the sqlite3 verion in the CI job does not support FTS5, but I'm not sure.
Expect this to take some time ...

Bouni, 2024-04-19:
OK, I guess I've fixed it. Despite not sure what did the trick in the end. I use Ubuntu 22.04 instead of 20.04 now and I install sqlite3.

closed 2024-04-19 by Bouni

whmountains, 2024-04-19:
Thanks @Bouni !