Bouni/kicad-jlcpcb-tools #449: Impossible to search for short queries like "2k" or "33"

whmountains, 2024-04-24:
Since the FTS5 upgrade, I am not able to search for queries of two characters or less.  I believe this was intentional but I want to report that such queries are actually quite common.  I use them for multiple parts per project, especially in conjunction with the the filters set to only show parts in the basic parts catalog.  Currently I am forced to search for these passives on the jlcpcb website.

Any workarounds would be appreciated.

chmorgan, 2024-04-24:
Yeah this is a limit of fts5, indexing substrings so short would result in a huge database size, see:

https://www.sqlite.org/fts5.html

```
Substrings consisting of fewer than 3 unicode characters do not match any rows when used with a full-text query. If a LIKE or GLOB pattern does not contain at least one sequence of non-wildcard unicode characters, FTS5 falls back to a linear scan of the entire table.
```

We *could* drop shorter tokens to avoid breaking the search at least as a short term fix.

It appears possible to fall back to a 'like', trying to test here but sqlitebrowser is crashing. I'll try on the command line in a few.

whmountains, 2024-04-25:
Ah!  That is where the limit came from.  I was looking all over the code but couldn't find any `length < 3` expression.

I'm not sure what to say.  I want FTS5 to work, since the search as you type experience is really nice.  But not being able to search for short strings is a workflow breaker for me.  Hopefully we can find some workaround.

If we could combine traditional indices with FTS5, then a fallback to a LIKE query would not be that slow in my work flow.  Generally I need short query support only with passives, and I will be adding other filters like footprint and unchecking the box for extended parts.  So it uses the index to narrow it down to a small list before scanning the description.  Only downside to this, if I understand correctly, is that it would require keeping a traditional parts table in addition to an FTS5 table similar to what was proposed in #444.

chmorgan, 2024-04-25:
@whmountains yeah just confirmed we can fall back. Let me get a PR for you to test tonight. Clearly it will be slower with the partial search but better than nothing. I'm assuming you'll be ok with a "AND description like '%33%'" term in the query? ie, you'll get matches on 330ohms, or 0.33mv etc.

I also think short searches like that are relatively rare. I actually don't think I've ever tried to search for something short. Too many things match. Now maybe I'd like to search for say "0603 10" to find 10V 0603s, but I'd probably keep typing and do "0603 10V".

whmountains, 2024-04-25:
OK, so you'd just type the package in the full-text search query?  That makes sense I guess.  I thought I tried that... I wonder what I did wrong.  Thanks for the clarification, I'll have to test it again.

Yeah in my case I would be fine typing 0603 33 to find a 33-ohm resistor for example.  I'm not sure how to type the ohm symbol so I revert to just scrolling through the list.  And generally I also filter to basic parts as well which helps to trim the list.

chmorgan, 2024-04-25:
@whmountains so I only type in the full text query, I never use any of the other selection boxes, it takes too much time to pick them. I just jam in the stuff I'd put into google, "0603 10uf 20V" etc and see what pops up.

Agree having the <3 character search is ideal for ohms. 

See #452

chmorgan, 2024-05-14:
Closing as fixed in #452

closed 2024-05-14 by chmorgan