Bouni/kicad-jlcpcb-tools #453: Plugin sometimes stops working after combining part files.

whmountains, 2024-04-25:
I've had this happen to me twice today while testing.  I'd say on my machine (MacOS 12.5.1, KiCad 8.0.1) it happens about 10% of the time I try to download a database.  What spurred me to open the issue was that I got the same problem while running in standalone mode, so I have a stack trace:

```
starting jlcpcbtools standalone mode...
2024.04.25 10:31:23 - DEBUG - download -  Parts db is split into 8 parts. Proceeding to download...
2024.04.25 10:31:23 - DEBUG - download -  Removing any spurios old zip part files...
2024.04.25 10:31:23 - DEBUG - download -  Removing /Users/me/Documents/KiCad/8.0/scripting/plugins/bouni-kicad-tools/jlcpcb/parts-fts5.db.zip.001.
2024.04.25 10:31:23 - DEBUG - download -  Removing /Users/me/Documents/KiCad/8.0/scripting/plugins/bouni-kicad-tools/jlcpcb/parts-fts5.db.zip.002.
2024.04.25 10:31:23 - DEBUG - download -  Removing /Users/me/Documents/KiCad/8.0/scripting/plugins/bouni-kicad-tools/jlcpcb/parts-fts5.db.zip.003.
2024.04.25 10:31:23 - DEBUG - download -  Download parts db chunk 1 with a size of 76.29MB
2024.04.25 10:31:25 - DEBUG - download -  Download parts db chunk 2 with a size of 76.29MB
2024.04.25 10:31:28 - DEBUG - download -  Download parts db chunk 3 with a size of 76.29MB
2024.04.25 10:31:30 - DEBUG - download -  Download parts db chunk 4 with a size of 76.29MB
2024.04.25 10:31:32 - DEBUG - download -  Download parts db chunk 5 with a size of 76.29MB
2024.04.25 10:31:34 - DEBUG - download -  Download parts db chunk 6 with a size of 76.29MB
2024.04.25 10:31:37 - DEBUG - download -  Download parts db chunk 7 with a size of 76.29MB
2024.04.25 10:31:39 - DEBUG - download -  Download parts db chunk 8 with a size of 17.90MB
2024.04.25 10:31:39 - DEBUG - download -  Combining and extracting zip part files...
Traceback (most recent call last):
  File "/Users/me/Documents/KiCad/8.0/scripting/plugins/bouni-kicad-tools/mainwindow.py", line 562, in populate_footprint_list
    if not self.store:
AttributeError: 'JLCPCBTools' object has no attribute 'store'
```

NOTE: the GUI remains responsive, there is no beachball, but I never get the success message.  

The referenced line contains the following code:
```python
if not self.store:
    self.init_store()
```

Smells like a race condition.  I don't know anything about how python wxwidgets interacts with multithreading, but I believe the error must be something along the lines of this: https://stackoverflow.com/a/31701465

whmountains, 2024-04-25:
Update, sometimes the success message does manage to show up in spite of the exception.

Bouni, 2024-07-04:
Is this still an issue or has #455 fixed this as well?

chmorgan, 2025-03-29:
Closing due to inactivity.

closed 2025-03-29 by chmorgan