Bouni/kicad-jlcpcb-tools #454: Crash in parts download due to "NSRangeException"

whmountains, 2024-04-25:
Full log is below:

```
/Users/kicad/remoteroot/workspace/build-macos-kicad-release/kicad-mac-builder/build/kicad/src/kicad/pcbnew/action_plugin.cpp(163): assert "PgmOrNull()" failed in register_action().
starting jlcpcbtools standalone mode...
2024.04.25 11:07:27 - INFO - setup -  Data directory 'jlcpcb' does not exist and will be created.
2024.04.25 11:07:27 - DEBUG - create_rotation_table -  Create SQLite table for rotations
2024.04.25 11:07:27 - DEBUG - download -  Parts db is split into 8 parts. Proceeding to download...
2024.04.25 11:07:27 - DEBUG - download -  Removing any spurios old zip part files...
2024.04.25 11:07:27 - DEBUG - download -  Download parts db chunk 1 with a size of 76.29MB
2024.04.25 11:07:29 - DEBUG - download -  Download parts db chunk 2 with a size of 76.29MB
2024.04.25 11:07:31 - DEBUG - download -  Download parts db chunk 3 with a size of 76.29MB
2024.04.25 11:07:34 - DEBUG - download -  Download parts db chunk 4 with a size of 76.29MB
2024.04.25 11:07:36 - DEBUG - download -  Download parts db chunk 5 with a size of 76.29MB
2024.04.25 11:07:38 - DEBUG - download -  Download parts db chunk 6 with a size of 76.29MB
2024.04.25 11:07:40 - DEBUG - download -  Download parts db chunk 7 with a size of 76.29MB
2024-04-25 11:07:40.933 Python[35252:7956581] *** Terminating app due to uncaught exception 'NSRangeException', reason: 'NSMutableRLEArray insertObject:range:: Out of bounds'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff80c1177c3 __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff80be77bc3 objc_exception_throw + 48
	2   Foundation                          0x00007ff80cf0559a -[NSMutableRLEArray insertObject:range:] + 786
	3   Foundation                          0x00007ff80cf17f14 -[NSMutableRLEArray replaceObjectsInRange:withObject:length:] + 777
	4   UIFoundation                        0x00007ff80f9880ce -[NSLayoutManager addTemporaryAttributes:forCharacterRange:] + 444
	5   AppKit                              0x00007ff80efaef87 -[NSTextView(NSTextCheckingClient) _addAnnotations:range:textStorage:removeExistingAnnotations:] + 1372
	6   AppKit                              0x00007ff80ece75f1 -[NSTextView(NSTextCheckingClient) setAnnotations:range:] + 196
	7   Foundation                          0x00007ff80cf057ae -[NSAttributedString enumerateAttributesInRange:options:usingBlock:] + 288
	8   AppKit                              0x00007ff80ece74a6 -[NSTextCheckingController _setAnnotatedSubstring:forRange:offset:replacementRanges:resultingRanges:] + 565
	9   AppKit                              0x00007ff80ece704d __48-[NSTextCheckingController insertedTextInRange:]_block_invoke + 2166
	10  AppKit                              0x00007ff80ec66f70 -[NSTextCheckingController annotatedSubstringForProposedRange:wrap:completionHandler:failureHandler:] + 874
	11  AppKit                              0x00007ff80ec66c00 -[NSTextCheckingController annotatedSubstringForProposedRange:completionHandler:] + 84
	12  AppKit                              0x00007ff80ece67cc -[NSTextCheckingController insertedTextInRange:] + 120
	13  AppKit                              0x00007ff80ece303f -[NSTextView _insertText:replacementRange:] + 3289
	14  libwx_osx_cocoau-3.2.0.2.2.dylib    0x000000010b660645 _ZN19wxNSTextViewControl9WriteTextERK8wxString + 229
	15  libwx_osx_cocoau-3.2.0.2.2.dylib    0x000000010b5d88ea _ZN11wxTextEntry9WriteTextERK8wxString + 90
	16  _core.cpython-39-darwin.so          0x0000000110d331db _ZL26meth_wxTextEntry_WriteTextP7_objectS0_S0_ + 155
	17  Python                              0x0000000105455855 cfunction_call + 69
	18  Python                              0x0000000105411af8 _PyObject_MakeTpCall + 376
	19  Python                              0x00000001054ff642 call_function + 626
	20  Python                              0x00000001054fc85e _PyEval_EvalFrameDefault + 28734
	21  Python                              0x0000000105412283 function_code_fastcall + 163
	22  Python                              0x00000001054ff56d call_function + 413
	23  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	24  Python                              0x0000000105412283 function_code_fastcall + 163
	25  Python                              0x00000001054ff56d call_function + 413
	26  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	27  Python                              0x0000000105412283 function_code_fastcall + 163
	28  Python                              0x00000001054ff56d call_function + 413
	29  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	30  Python                              0x0000000105412283 function_code_fastcall + 163
	31  Python                              0x00000001054ff56d call_function + 413
	32  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	33  Python                              0x00000001055004a6 _PyEval_EvalCode + 2950
	34  Python                              0x00000001054121cd _PyFunction_Vectorcall + 253
	35  Python                              0x00000001054146f2 method_vectorcall + 514
	36  Python                              0x00000001054fcd45 _PyEval_EvalFrameDefault + 29989
	37  Python                              0x00000001055004a6 _PyEval_EvalCode + 2950
	38  Python                              0x00000001054121cd _PyFunction_Vectorcall + 253
	39  Python                              0x00000001054ff56d call_function + 413
	40  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	41  Python                              0x0000000105412283 function_code_fastcall + 163
	42  Python                              0x0000000105414663 method_vectorcall + 371
	43  Python                              0x00000001054fcd45 _PyEval_EvalFrameDefault + 29989
	44  Python                              0x0000000105412283 function_code_fastcall + 163
	45  Python                              0x00000001054ff56d call_function + 413
	46  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	47  Python                              0x0000000105412283 function_code_fastcall + 163
	48  Python                              0x00000001054ff56d call_function + 413
	49  Python                              0x00000001054fc831 _PyEval_EvalFrameDefault + 28689
	50  Python                              0x0000000105412283 function_code_fastcall + 163
	51  Python                              0x0000000105414663 method_vectorcall + 371
	52  Python                              0x00000001055a3227 t_bootstrap + 71
	53  Python                              0x00000001055518f9 pythread_wrapper + 25
	54  libsystem_pthread.dylib             0x00007ff80bfd54e1 _pthread_start + 125
	55  libsystem_pthread.dylib             0x00007ff80bfd0f6b thread_start + 15
)
libc++abi: terminating with uncaught exception of type NSException
```

This occured *before* the zip files can be merged, but after all the plugins have downloaded.  May or may not be related to #453 and #448.  The symptoms and timing is quite different, but if there is some fundamental thread unsafety or memory corruption then it's not surprising the errors would occur semi-randomly.

chmorgan, 2024-04-26:
Can you see if #455 fixes the issue for you? I'm able to reproduce the error here (64GB of ram etc) before the change but not after it.

whmountains, 2024-04-30:
Closing since I believe #455 will fix it.  If it does not we can always re-open.

closed 2024-04-30 by whmountains