Bouni/kicad-jlcpcb-tools #456: Rotations empty on KiCAD 8.0

netkruzer, 2024-04-27:
Rotations are empty:
[image]

**Describe the bug**
All rotations are empty. I used this plugin with KiCAD 7.x and it exported the rotations great. In 8.0 it no longer exports or shows the fixed rotations. Importing into JLCPCB's assembly menu, all rotations are off. 

[image]
These same rotation mappings worked for me in KiCAD 7.x

JLCPCB Tools version 2024.04.03

**Expected behavior**
Should have rotations

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 8.0.1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 15 2024 01:52:47
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:

```

lk-davidegironi, 2024-05-10:
Same problem to me on KiCad 7.0.11 on Win 10, JLCPCB Tools version 2024.04.03

[image]

Bouni, 2024-05-13:
Fixed in #445

closed 2024-05-13 by Bouni